markqvist/RNode_Firmware #10: SX1262 support

tomasbrincil, 2021-09-23:
Hi there, I really like this project.
I am interested in SX1262. I was checking LoRa class and you did very hard work there.
How difficult would be to implement another RD module?
Regards!

markqvist, 2021-09-30:
Support for SX1262 is coming in the relatively near future. Stay tuned :)

markqvist, 2021-10-05:
Closing this for now, as it's not an issue as such. You are still welcome to ask any follow-up questions here though.

tomasbrincil, 2022-05-19:
Hi Mark,
I see a huge improvment there. Still I would like to use sx1262. From code it looks like sx1276 only is supported.
Original LoRa.c from Sandeep Mistry works only with 127X family. I guess due to simplicity you dont want use all heavy RadioLib library but have you considered some solution for sx1262 support? Thank you! I can help with a code, or support hardware or kofi.

markqvist, 2022-05-28:
SX1262 support is coming. I will release more info soon, but I am actively working on this. If you have any specific boards you would like to see supported, would you mind listing them here? Also, hardware donations in this direction are very much appreciated, and will help definitely help out :)

markqvist, 2022-05-28:
Actually, I will reopen this, just so there is visibility about it for others with the same interest.

khimaros, 2022-10-16:
i'm also trying to setup an RNode on my TTBeam 1.1.

this is the specific board that i have:

https://www.aliexpress.us/item/2255801100907218.html?spm=a2g0o.order_list.0.0.58091802dPVCTN&gatewayAdapt=glo2usa4itemAdapt&_randl_shipto=US

markqvist, 2022-10-19:
Thanks for the report @khimaros, and thanks for the resources. Meshtastic uses radiolib to actually handle interfacing with the LoRa modules (as far as I remember) whereas the RNode firmware uses it's own custom driver directly (since we need a really tight and interrupt-driven implementation to get the kind of performance we need, especially in terms of CSMA performance). This means that I need to write a new driver for SX1262 chips.

It's not really a big deal, and not even really a big project as such, I simply have not had time to complete it yet, because other things have been prioritised higher.

At the (extremely low) level of funding and resources I currently have, things just have to be prioritised tightly in terms of what is providing most real-world benefit at any given point.

Hopefully we can add SX1262 (and family) support soon. You are very welcome to help out with donations!

markqvist, 2022-10-25:
I received a donation to cover hardware expenses related to implementing SX1262/68 support. The boards are now ordered and should be here soon.

markqvist, 2022-11-03:
OK, so devices came in, and of course they have the *old chips*. Great. I will order some new ones and hope I get lucky this time around. If anyone has a reputable source for devices with SX1262 chips (T-Beams for example), please let me know.

madbird99, 2022-11-11:
[AliExpress](https://www.aliexpress.com/)
[CDSENET Official Store](https://www.aliexpress.com/store/2798183?spm=a2g0o.detail.1000007.1.f9ac4f62xhWm54)

markqvist, 2022-11-12:
Thanks @madbird99, but I'm looking specifically for T-Beam boards with the SX1262/SX1268 chips, not raw modules.

markqvist, 2022-11-12:
Ah wait, I did find some there now. Last time I looked I could only find them with SX1276 chips, or listings that were unclear on what chip was used.

markqvist, 2022-11-12:
Ok, new boards ordered. Hopefully they have the SX1262/SX1268 chips this time around :)

khimaros, 2023-01-22:
@markqvist if you still have trouble getting your hands on the right chip after the latest, i'd be happy to send you one of mine.

markqvist, 2023-01-22:
Thanks @khimaros! It took a while, but the boards I ordered finally arrived a short while ago, and this time around, they do actually have the SX1262 chips onboard ;) So I think I'm covered for now, but thanks a lot for the offer!

IhorNehrutsa, 2023-05-23:
@markqvist 
Hi, Mark.
May you open a branch (pull request) with the current SX1262/SX1268 development?

n4ijk, 2023-07-13:
I would like to offer what ever help is needed to get the heltec V3 supported. I have two of these from a meshtastic project I'm not using and would love to implement them for this.

markqvist, 2023-07-13:
> I would like to offer what ever help is needed to get the heltec V3 supported. I have two of these from a meshtastic project I'm not using and would love to implement them for this.

Thanks! Right now the biggest obstacle is creating a general-purpose driver for the SX1262 chip, similar to the one the RNode firmware currently uses for the SX1276. If you're up for it, that's where to start!

n4ijk, 2023-07-14:
I'm not a very hood at this type of development, but I'll give it a shot this weekend.

keidrych, 2023-12-31:
G'Day @markqvist seasons greetings, just seeing if work on the SX1262 chips is still moving forward, do you need any help or assistance?

jacobeva, 2024-01-21:
This work is now complete, this issue can be closed.

markqvist, 2024-01-21:
Implemented by #55.

closed 2024-01-21 by markqvist