markqvist/RNode_Firmware #11: RNode for ESP32 using ESP-IDF

HarlekinSimplex, 2021-10-05:
Is there a way to port this code to ESP-IDF to run RNode on a TTGO T-Beam or like ESP32 based LoRa device?
Cheers
Stephan

markqvist, 2021-10-05:
This is something I am working on, and will release more details about after the next release of NomadNet and Reticulum. Since this is not directly an issue with the RNode firware, I'm closing this. You're welcome to use this issue to ask any further questions on the subject though.

markqvist, 2021-10-05:
Also, I would be curious to hear about your use-case for this. Would you mind detailing it?

closed 2021-10-05 by markqvist

HarlekinSimplex, 2021-10-05:
Sure.

We try to build and deploy an emergency radio network at our community.

Initially we opted to use Meshtastic and LoRa as the primary means of communication.

After a while we found out that Mesht has issues with lots of channels and broad coverage of large areas. For that we need to interconnect with other communication technologies.

Now our tech team decided to switch to RNode and Reticulum.

However, our friends around already have purchased quit a lot ESP32 T-Beams for the Meshtastic approach that we don’t want to trash. Further we like to have the Wifi Options and other stuff that’s available on the ESP-IDF platform for the ESP32 as well to do other things with the T-Beams if we so choose.

So I started searching for a way to port the RNode code to ESP-IDF so that we can use the ESP32 via the USB or WiFi as LoRa Device for Reticulum.

Cheers

Stephan

tomasbrincil, 2021-10-05:
Sorry for opening this. But if you check my “issue” with sx1262 it was basically because of tons of tbeams v1.1

markqvist, 2021-10-06:
Ah, I understand. Thanks for indulging my curiosity. As hinted above, the current RNode firmware does not port directly to ESP32, since there's way too many low-level calls tied directly to the assumptions of the AVR platform. But there's nothing hindering an ESP32 firmware that is compatible with Reticulum.

The one I am working on now will have modular support for a range of transceivers, including both the SX1262, SX1268, SX1276 and friends in both LoRa and FSK, MSK and GMSK modes, WiFi and Bluetooth, and offer proper multiple access schemes. I should be able to squeeze in a software-defined modem for HF/VHF/UHF radios as well.

markqvist, 2022-01-19:
There is now initial support for T-Beam devices in the RNode firmware. You can install the RNode firmware automatically using "rnodeconf --autoinstall". Let me know if there's any questions :)