markqvist/RNode_Firmware #17: (feature request) Support for Heltec LoRa32 V2.1

RicInNewMexico, 2022-04-23:
Would love support for the Heltec LoRa32 V2.1 boards. From what I can see at initial glance it shouldn't take much to port over. The board is an ESP32-DOWDQ6 MCU with an SX1276 SPI interface LORA chipset.

In regard to the LORA chipset pinout it's very similar to the Lilygo LoRa32 V2.1 board which is already supported except it seems that the reset line is bound to pin 14 instead of pin 23 used by the Lilygo board. 

Thoughts? I'd be willing to supply a couple boards for development if needed. Cheapest US-based source I've found for them is Amazon. Although [THIS](https://www.amazon.com/Development-Antenna-Arduino-LoraWan-Internet/dp/B07HJ49VN8) Amazon listing pictures the older version and doesn't have them listed as Heltec brand boards they **are** shipping the current Heltec LoRa32 v2.1 boards (I've purchased twice so far working on development of another project and had no complaints).

markqvist, 2022-04-26:
If you are willing donate a couple of them to the cause and send them to me, I'll add support for them to the tools (if at all possible, which I am 99% sure is no problem).

RicInNewMexico, 2022-04-26:
Absolutely, what would be the best method for doing so? Would you prefer I order a couple extra and re-ship them to you or would it be easier if I made a ko-fi donation to cover the cost and you order them directly?

markqvist, 2022-04-27:
Thanks, that is great man! The easiest is probably if you just make a donation on ko-fi, then I can order them for direct shipment. I found them on amazon for 32eur a piece, so if you donate 64, I can buy a pair.

kc1awv, 2022-06-12:
Not sure how for you've gotten on this, but I can say with some slight confidence that the firmware works here. Changing the values for the pins in BOARD_GENERIC_ESP32 and compiling for that target results in a firmware that's uploaded and looks as if it's running. There is only one LED that can be controlled on the board (pin 25), the other is a power LED in line with the +5V rail. I substituted a random 'free' pin for the RX LED.

Config.h:
`#define BOARD_MODEL BOARD_GENERIC_ESP32

#if BOARD_MODEL == BOARD_GENERIC_ESP32
	const int pin_cs = 18;
	const int pin_reset = 14;
	const int pin_dio = 26;
	const int pin_led_rx = 39;
	const int pin_led_tx = 25;`

And, reading using rnodeconfig:
`[smiller@eifersucht ~]$ rnodeconf /dev/ttyUSB0 -i
[2022-06-12 14:41:02] Opening serial port /dev/ttyUSB0...
[2022-06-12 14:41:05] Device connected
[2022-06-12 14:41:05] Current firmware version: 1.29
[2022-06-12 14:41:05] Reading EEPROM...
[2022-06-12 14:41:05] EEPROM is invalid, no further information available`

Are you still looking for funds to purchase these boards? If so, I can help.

kc1awv, 2022-06-12:
Ok, after fudging and forking a bit, I was able to load up and initialize the Heltec WiFi LoRa 32 v2 board.

[2022-06-12 17:41:22] Opening serial port /dev/ttyUSB0...
/home/smiller/git/rnodeconfigutil/rnodeconf/./rnodeconf.py:1475: DeprecationWarning: setDaemon() is deprecated, set the daemon attribute instead
  thread.setDaemon(True)
[2022-06-12 17:41:25] Device connected
[2022-06-12 17:41:25] Current firmware version: 1.29
[2022-06-12 17:41:25] Reading EEPROM...
[2022-06-12 17:41:26] EEPROM checksum correct
[2022-06-12 17:41:26] Device signature validated
[2022-06-12 17:41:26] 
[2022-06-12 17:41:26] Device info:
[2022-06-12 17:41:26] 	Product            : Hombrew RNode (Band capabilities unknown) (f0:ff:35)
[2022-06-12 17:41:26] 	Device signature   : Validated - Local signature
[2022-06-12 17:41:26] 	Firmware version   : 1.29
[2022-06-12 17:41:26] 	Hardware revision  : 1
[2022-06-12 17:41:26] 	Serial number      : 00:00:00:01
[2022-06-12 17:41:26] 	Frequency range    : 100.0 MHz - 1100.0 MHz
[2022-06-12 17:41:26] 	Max TX power       : 14 dBm
[2022-06-12 17:41:26] 	Manufactured       : 2022-06-12 17:40:57
[2022-06-12 17:41:26] 	Device mode        : Normal (host-controlled)

markqvist, 2022-06-13:
Hey @kc1awv! Thanks for looking into this, and good to hear that it looks positive initially. @RicInNewMexico didn't get back to this after the initial proposal, so this thread have actually just been sitting here, and I didn't get any further.

If anyone is still willing to sponsor a couple of units, I will be happy to add the support officially. I just need to verify that it is all working as it should, add to the correct pin-mappings and any tweaks/LED mappings, and then assign a model/board code to the config tools, so it can be recognized properly for firmware updates and such.

So yeah, if anyone is up for covering the cost of the units, I will order them straight away and get support added in no time :)

kc1awv, 2022-06-13:
> Also, did you have a chance to test it on-air @kc1awv?

Not exactly... Reticulum had issues trying to open the Heltec device. Might be something to do with firmware signing? Not sure. I switched over to a couple T-Beams I had lying around, just to get NomadNet up and running locally.

I'll send some funds to you to help. I can test locally as well, I've a handful of the 900MHz Heltec devices waiting to be put to use.

markqvist, 2022-06-13:
Cool, thank you so much. I'll get them working :)

markqvist, 2022-06-16:
Boards came in, should have them supported shortly!

[image]

markqvist, 2022-06-16:
And it's ready! Both the firmware (version 1.29) and `rnodeconf` (version 1.2.1) now supports the Heltec boards. Let me know if you run into any issues.

Thanks for the support in adding this board!

closed 2022-06-16 by markqvist