markqvist/RNode_Firmware #19: Repeater shift (separate TX and RX frequencies)

ivanpiter, 2022-05-21:
Please, add setup for repeater shift (separate TX and RX frequencies).

tomasbrincil, 2022-05-23:
Yes, I am into split mode too.

markqvist, 2022-05-23:
Thanks for the suggestion! I actually already had this on my list of possible features, so good to know there is interest in it!

markqvist, 2022-05-28:
This is coming in the future :)

tomdelta, 2022-06-28:
Yes, I need it also !

Tom

markqvist, 2022-06-28:
It will be included in the v2 release of the firmware. There us more info on that here: https://unsigned.io/rnode-a-new-model/

whitecd, 2022-07-27:
I agree mixed Rx/Tx characteristics could be useful as a low-power repeater/bridge router.

As to V2 release, may I suggest using the Zephyr OS (https://zephyrproject.org/) as dev environment.  Zephyr is a Linux Foundation embedded ROS project. It was developed upfront with security and highly constrained SBCs.  Zephyr would supports a large number of SBCs including most of the existing RNode boards. Those not currently supported can easily be added.  There is a GSOC effort to support the MBED ardruino framework under Zephyr.

whitecd, 2022-08-24:
There is a new (GSOC 2022) project ( https://github.com/zephyrproject-rtos/gsoc-2022-arduino-core) that may assist in a "quick" port of the existing V1.x Arduino framework to Zephyr while you work a native V2 port.  This GSOC effort is a port of MBED's Arduino framework into Zephyr.  I don't know how robust it is to date, but it supposedly works at least at an alpha stage.
Zephyr will give you robust OTA FW upgrades (MCUBoot) as well as robust radio support/abstraction.

markqvist, 2022-08-25:
Thanks for bringing that up, I will look into it!

markqvist, 2024-02-29:
No takers for implementation. Closing as out of scope for now.

closed 2024-02-29 by markqvist