markqvist/RNode_Firmware #20: RNode TXpower setting

tomdelta, 2022-06-28:
Hello Mark,

I made some different homemade RNode. Rnode 169MHz basend on ATmega1284p+ RFM95PW-169, a new one 169Mhz based on lolin ESP32 and 868MHz based on T-Beam. Regardles what I set in the reticulum config (0-17dBm) at the 169 rnodes I got 19dBm output and on the TBeam 0dBm output power. I updated rnodeconfig and the rnodes today. I tested also TNC mode with APRSdroid and with examply.py with different powersettings, same result! Did I something wrong? I you want I can send you some of the 169MHz/500mW Lolin RNodes for testing.

Best regards
Tom

markqvist, 2022-06-28:
Hi Tom
Could you clarify whether the RNodes report back the wrong tx power, or whether they report the right tx power, but measurement equipment tells the actual tx power is different, or they report and output a different tx power than set? There has been one otjer report of something similar, and I have not yet been able to reproduce the issue, unfortunately.

tomdelta, 2022-06-29:
Hi Mark,
RNode reports back the setting I made! But no change in output power.

ok mit dem TNC set ging es wohl

$ rnodeconf /dev/ttyUSB1 -T
[2022-06-28 22:50:54] Opening serial port /dev/ttyUSB1...
[2022-06-28 22:50:58] Device connected
[2022-06-28 22:50:58] Current firmware version: 1.28
[2022-06-28 22:50:58] Reading EEPROM...
[2022-06-28 22:50:58] EEPROM checksum correct
[2022-06-28 22:50:58] Device signature validated
[2022-06-28 22:50:58] Please input startup configuration:

Frequency in Hz: 867200000
Bandwidth in Hz: 125000
TX Power in dBm: 14
Spreading factor: 10
Coding rate:  5

[2022-06-28 22:52:04] Radio reporting frequency is 867.2 MHz
[2022-06-28 22:52:04] Radio reporting bandwidth is 125.0 KHz
[2022-06-28 22:52:04] Radio reporting TX power is 14 dBm
[2022-06-28 22:52:04] Radio reporting spreading factor is 10
[2022-06-28 22:52:04] Radio reporting coding rate is 5
[2022-06-28 22:52:05] Device set to TNC operating mode
tom@mx:~
$ rnodeconf /dev/ttyUSB1 -i
[2022-06-28 22:52:15] Opening serial port /dev/ttyUSB1...
[2022-06-28 22:52:18] Device connected
[2022-06-28 22:52:18] Current firmware version: 1.28
[2022-06-28 22:52:18] Reading EEPROM...
[2022-06-28 22:52:19] EEPROM checksum correct
[2022-06-28 22:52:19] Device signature validated
[2022-06-28 22:52:19] 
[2022-06-28 22:52:19] Device info:
[2022-06-28 22:52:19]  Product            : LilyGO T-Beam 850 - 950 MHz (e0:e9:33)
[2022-06-28 22:52:19]  Device signature   : Validated - Local signature
[2022-06-28 22:52:19]  Firmware version   : 1.28
[2022-06-28 22:52:19]  Hardware revision  : 1
[2022-06-28 22:52:19]  Serial number      : 00:00:00:25
[2022-06-28 22:52:19]  Frequency range    : 850.0 MHz - 950.0 MHz
[2022-06-28 22:52:19]  Max TX power       : 17 dBm
[2022-06-28 22:52:19]  Manufactured       : 2022-06-28 19:44:11
[2022-06-28 22:52:19] 
[2022-06-28 22:52:19]  Device mode        : TNC
[2022-06-28 22:52:19]    Frequency        : 867.2 MHz
[2022-06-28 22:52:19]    Bandwidth        : 125.0 KHz
[2022-06-28 22:52:19]    TX power         : 14 dBm (25.119 mW)
[2022-06-28 22:52:19]    Spreading factor : 10
[2022-06-28 22:52:19]    Coding rate      : 5
[2022-06-28 22:52:19]    On-air bitrate   : 0.98 kbps

tomdelta, 2022-06-29:
[image]

markqvist, 2022-06-29:
Thanks for providing the details Tom! I have updated the precompiled builds.

Could you try updating your RNodes with `rnodeconf -u`? They should report version 1.29 then, and have correct output power (although they will still not take in to account the apmlifier stage, of course).

Also, I could not find any info on how the RF path is wired to the SX1276 RF_OUTPUT and/or PA_BOOST pins for the 169 MHz modules you are using, so I took a semi-educated guess ;) Let me know if it works!

It would be great if you could send me a couple of each unit to test with!

markqvist, 2022-06-29:
Ah great, they must have the output wired differently than what I assumed... Well, there is not too many options, so maybe we should just try it out. Give me a sec...

tomdelta, 2022-06-29:
[image]

tomdelta, 2022-06-29:
For rx pin 1 is used rfin_lf

markqvist, 2022-06-29:
Thanks, that's what is wrong then, I assumed it would be the RFO pin. How are you setting them up? Are you specifying them as homebrew RNodes in the installer for both of them?

tomdelta, 2022-06-29:
Yes homebrew ESP32 for the lolin32 169 MHz. For the 868 I used lilygo

tomdelta, 2022-06-29:
[image]

markqvist, 2022-06-29:
Thanks Tom, I think I got it set up correctly for your configuration now, but I needed to add an option to the autoinstaller, which I am doing now... Update incoming.

tomdelta, 2022-06-29:
Thanks! Regarding the New rnode V2 we are surely interested! We have very good offers from hoperf for large amount of moduls. Maybe we can work together. I ll send you PM . MY adapterboard can hold two modules. Would IT ne possible to use two Lora with different adresses  on one rnode? Then we could have 169/868 on one rnode. If the rns can work on the rnode as repeater IT would bei perfekt as stand alone router

markqvist, 2022-06-29:
Ok, I just uploaded version 1.2.2 of rnodeconf, and version 1.30 of the RNode firmware. Try that one out, and please let me know how it goes. There is an extra step in the auto-installer for selecting the RF output pin for homebrew RNodes, and the output is now corrected back to working order, plus I fixed some register stuff that prevented the lilygo board from actually changing the output.

Yes, the V2 firmware supports handling multiple different transceivers on the same device. Very handy. Saw your message, so let's continue the discussion about that there.

Let me know if it works now!

tomdelta, 2022-06-29:
[image]

tomdelta, 2022-06-29:
27dbm output with setting 14dbm in config testen with example.py

tomdelta, 2022-06-29:
I ll test now the 868 RNode and the I try to setup a attenuation link over cable so see the possible link budge.

markqvist, 2022-06-29:
Looking forward to hearing how it works out!

markqvist, 2022-06-29:
I'll close this now, as the issue itself is resolved, but feel free to post more here if anything else crops up.

closed 2022-06-29 by markqvist

markqvist, 2022-06-29:
Sweet stuff! That is great to hear man! Thanks for reporting back. Interesting that it actually goes that far down.

Is the spectrum analyzer you are using presenting a 50 ohm load? We could expect it to actually radiate effectively on the 433 band as well then...

tomdelta, 2022-06-29:
Yes sure! It was measured with a 30dB attenuator 50ohm but also the specy have a 50ohm input but max. 20dBm. I ll try to transmit with it at a real antenna.

markqvist, 2022-06-29:
That is very interesting. I did some tests a long time ago with some other modules, but they were not very promising. If both reception and transmission works down to 365 MHz, and probably up to around 1GHz, that would be very interesting and useful. Looking forward to hearing your results.