markqvist/RNode_Firmware #22: Building for NRF52

samuk, 2022-10-22:
Has anyone tried to build for NRF52 devices eg https://www.aliexpress.com/item/1005003652461302.html or would they not be powerful enough. They are very power efficient.

markqvist, 2022-10-22:
Interesting SoC. It can definitely run the RNode firmware, there just isn't a port for it yet.

Since it is based on an ARM Cortex-M4, it would entail adding a new platform to the build system, which is a bit of work in the current form of the firmware, so I don't think support for this will be until the v2 branch of the firmware is released.

There is also the pending task of [adding a driver for SX1262/68](https://github.com/markqvist/RNode_Firmware/issues/10).

But yeah, totally doable, just not there yet :)

samuk, 2022-10-22:
I think this would also support https://store.rakwireless.com/products/wisblock-meshtastic-starter-kit for those that want a cheaper/modular thing.

Both Ttgo & RAK may pre-flash devices if it seems like you'll sell decent volumes.

jacobeva, 2024-01-21:
I should think that this issue should be closed now :)

markqvist, 2024-01-21:
Implemented by #55. Thanks to @jacobeva for making this happen :)

closed 2024-01-21 by markqvist