markqvist/RNode_Firmware #24: Autoconfig error

megabyte2010, 2022-11-03:
Autoconfig can't install because can't find firmware in github releases when we specify --fw specific firmware. Using the menu options, he says that there isn't internet connection. I'm not at home, so I can't copy paste the error but it's something like this. Thank you.

markqvist, 2022-11-03:
What specific firmware version are you trying to install?

megabyte2010, 2022-11-03:
rnode_firmware_lora32v21

megabyte2010, 2022-11-03:
PS C:\Users\acer> rnodeconf --autoinstall

Hello!

This guide will help you install the RNode firmware on supported
and homebrew devices. Please connect the device you wish to set
up now. Hit enter when it is connected.

Detected serial ports:
  [1] COM3 (None, 02061D89)

What serial port is your device connected to? 1

Ok, using device on COM3 (None, 02061D89)

Probing device...
[22:53:10] No answer from device

It looks like this is a fresh device with no RNode firmware.
What kind of device is this?

[1] RNode from Unsigned.io
[2] Homebrew RNode
[3] LilyGO T-Beam
[4] LilyGO LoRa32 v2.0
[5] LilyGO LoRa32 v2.1
[6] Heltec LoRa32 v2

? 5

---------------------------------------------------------------------------
Important! Using RNode firmware on LoRa32 devices should currently be
considered experimental. It is not intended for production or critical use.
The currently supplied firmware is provided AS-IS as a courtesey to those
who would like to experiment with it. Hit enter to continue.
---------------------------------------------------------------------------

What band is this LoRa32 for?

[1] 433 MHz
[2] 868 MHz
[3] 915 MHz
[4] 923 MHz

? 2

Ok, that should be all the information we need. Please confirm the following
summary before proceeding. In the next step, the device will be flashed and
provisioned, so make that you are satisfied with your choices.

Serial port     : COM3
Device type     : LilyGO LoRa32 v2.1 850 - 950 MHz
Platform        : ESP32
Device MCU      : Espressif Systems ESP32
Firmware file   : rnode_firmware_lora32v21.zip

Is the above correct? [y/N] y
[22:53:29] Checking firmware file availability...
[22:53:29] Failed to retrive latest version information for your board.
[22:53:29] Check your internet connection and try again.
[22:53:29] If you don't have Internet access currently, use the --fw-version option to manually specify a version.
PS C:\Users\acer>

megabyte2010, 2022-11-03:
With the --fw option the behaviour is:

Serial port     : COM3
Device type     : LilyGO LoRa32 v2.1 850 - 950 MHz
Platform        : ESP32
Device MCU      : Espressif Systems ESP32
Firmware file   : rnode_firmware_lora32v21.zip

Is the above correct? [y/N] y
[22:55:51] Checking firmware file availability...
[22:55:51] Downloading missing firmware file: rnode_firmware_lora32v21.zip for version rnode_firmware_lora32v21.zip
[22:55:52] Could not download required firmware file:
[22:55:52] https://github.com/markqvist/RNode_Firmware/releases/download/rnode_firmware_lora32v21.zip/rnode_firmware_lora32v21.zip
[22:55:52] The contained exception was:
[22:55:52] HTTP Error 404: Not Found
PS C:\Users\acer>

megabyte2010, 2022-11-03:
I'm not seeing nothing in the releases folder. No zip files.

markqvist, 2022-11-03:
Ah, I actually have no idea if it will even work on Windows.

There's a lot of system calls that assume as POSIX OS, so I am not really expecting it to work on windows unless it is properly ported.

That specific failure you get there is probably because there is no `wget` utility available, so it cannot download the release info.

Can you try on a machine with Linux? It should even work on macOS out of the box too.

markqvist, 2022-11-06:
Moving this to a discussion, since it is not an issue relevant to the RNode firmware itself.