markqvist/RNode_Firmware #25: Issue with USB connection with V.152 on Heltec32v2

whitecd, 2022-11-09:
Upgraded Heltec32 to V.1.52 via "sudo chmod 777 /dev/ttyUSB0; rnodeconf /dev/ttyUSB0 --autoinstall".  This FW upgrade was successful.
When running Nomadnet or RNSD, I get a number of issue referencing either /dev/ttyUSb0 errors or inability to identify the LoRA radio:

[2022-11-09 12:49:06] [Notice] Serial port /dev/ttyUSB0 is now open
[2022-11-09 12:49:06] [Verbose] Configuring RNode interface...
[2022-11-09 12:49:06] [Verbose] Wating for radio configuration validation for RNodeInterface[RNode LoRa Interface]...
[2022-11-09 12:49:06] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting frequency is 902.3 MHz
[2022-11-09 12:49:06] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting bandwidth is 125.0 KHz
[2022-11-09 12:49:06] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting TX power is 7 dBm
[2022-11-09 12:49:06] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting spreading factor is 8
[2022-11-09 12:49:06] [Debug] RNodeInterface[RNode LoRa Interface] Radio reporting coding rate is 5
[2022-11-09 12:49:06] [Verbose] RNodeInterface[RNode LoRa Interface] On-air bitrate is now 3.12 kbps
[2022-11-09 12:49:07] [Notice] RNodeInterface[RNode LoRa Interface] is configured and powered up
[2022-11-09 12:49:07] [Notice] Reconnected serial port for RNodeInterface[RNode LoRa Interface]
[2022-11-09 12:49:07] [Error] A serial port error occurred, the contained exception was: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
[2022-11-09 12:49:07] [Error] The interface RNodeInterface[RNode LoRa Interface] experienced an unrecoverable error and is now offline.
[2022-11-09 12:49:07] [Error] Reticulum will attempt to reconnect the interface periodically.
[2022-11-09 12:49:12] [Verbose] Attempting to reconnect serial port /dev/ttyUSB0 for RNodeInterface[RNode LoRa Interface]...
[2022-11-09 12:49:12] [Notice] Opening serial port /dev/ttyUSB0...
[2022-11-09 12:49:12] [Verbose] Attempting to reconnect serial port /dev/ttyUSB0 for RNodeInterface[RNode LoRa Interface]...
[2022-11-09 12:49:12] [Notice] Opening serial port /dev/ttyUSB0...
[2022-11-09 12:49:14] [Error] A serial port error occurred, the contained exception was: ord() expected a character, but string of length 0 found

These errors are cyclic and repeat for the duration of the session.  Similar serial comm errors occur in -N and -T modes.
Please let me know if you need additional logs, etc.  I was also unable to downgrade the Heltec FW to see if I could regain use of the radio,  I have similar issues with all 3 Heltec radios with V.152, but did not have issues with earlier versions.
Thanks for your time and efforts.

markqvist, 2022-11-09:
Thanks for reporting this. I know it is probably not the cause, but just to be sure, could there be something else on the system trying to automatically access the serial port? Like a 3D-printer slicer, printer drivers, the Arduino IDE, or others?

Because those errors look exactly like what I have had on multiple occasions where I accidentially had CUPS try to use the same port, of left Slic3r open in the background.

markqvist, 2022-11-10:
Probably also worth mentioning here, is that I've noticed the power supplys on all the Heltec boards I have had are *real* finnicky. Use a just slightly under-spec USB cable, and they brown out and reset constantly. That *might* also be the cause of the above issue.

whitecd, 2022-11-10:
Sir,

I noticed weird errors with pip3 on Ubuntu 20.04, so I purged Python/pip3, rnsd, rnodeconf, and Nomadnet.  Once I reinstalled them, it appears that solved the problem on Ubuntu 20.04.  I had removed/reinstalled rinsed, rnodeconf, and Nomadnet previously but without much success.  I also ran “pip3 check” which did show a cffi as missing.  Installing that didn’t provide a fix- reinstalling Python/pip3 appears to.  I tried a similar approach for Ubuntu 22.04 and determined that removing Python craters the OS. I’m in the process of building a new 22.04 desktop and will let you know the outcome.
Good point about the power supply.  I did change USB cables out and no change.  Probably best in the future to run them with a LiON battery attached in order to mitigate the inrush effects, especially with BT activate.  Thanks for the info.
I noticed with the earlier version of Sideband that I needed to enable “UDP protocol” in RNSD in order to get have Sideband announcements heard on the LAN.  It appears that “Auto Interface’ did not provide the appropriate connectivity.  I have not confirmed that with the new Sideband/Nomadnet versions.
 
Question:
I need to acquire more RNode HW to support testing, what HW would you recommend given your direction in Rhode features being exposed.  The T-Echo has some nice features including GPS, but not currently supported.
How do I create a trusted LXMF propagation node?  I have enabled RNS node which I understand enables RNS path discovery/propagation, but not LXMF caching.  I assume NomadNet “node” mode provides LXMF caching, but I have not been able to make that work.  
Thanks,
Curt

markqvist, 2022-11-17:
Ok, let me know if you come up with anything interesting. Also check that your versions are at least:

- Reticulum/rnsd: `0.4.1`
- Nomadnet: `0.2.7`
- Sideband: `0.3.0`
- RNode firmware: `1.52`

It should not be necessary to enable any UDP interfaces for the announces to work over LAN/WiFi, if that is the case I think there is something wrong, either old versions of something, or firewall blocking the multicast peering packets.

markqvist, 2022-11-17:
Also, for hardware recommendations, I currently recommend this board: https://markqvist.github.io/Reticulum/manual/hardware.html#lilygo-lora32-v2-1

killabit, 2022-12-16:
I have a Heltex Lora32 V2 and i'm getting an error that says chil detected ESP32-S3 looking for ESP32, any ideas?

markqvist, 2022-12-16:
Can you post the full output of the error?

killabit, 2022-12-16:
"A fatal error occurred: This chip is ESP32-S3(beta3) not ESP32. Wrong --chip argument?"

Serial port     : /dev/ttyUSB0
Device type     : Heltec LoRa32 v2 850 - 950 MHz
Platform        : ESP32
Device MCU      : Espressif Systems ESP32
Firmware file   : rnode_firmware_heltec32v2.zip

Is the above correct? [y/N] y
[07:18:11] Checking firmware file availability...
[07:18:12] The latest firmware for this board is version 1.54
[07:18:12] Downloading missing firmware file: rnode_firmware_heltec32v2.zip for version 1.54
[07:18:14] Firmware file downloaded
[07:18:14] Veryfying firmware integrity...
[07:18:14] Extracting firmware...
Archive:  /root/.config/rnodeconf/update/1.54/rnode_firmware_heltec32v2.zip
  inflating: /root/.config/rnodeconf/update/1.54/esptool.py  
  inflating: /root/.config/rnodeconf/update/1.54/rnode_firmware_heltec32v2.boot_app0  
  inflating: /root/.config/rnodeconf/update/1.54/rnode_firmware_heltec32v2.bin  
  inflating: /root/.config/rnodeconf/update/1.54/rnode_firmware_heltec32v2.bootloader  
  inflating: /root/.config/rnodeconf/update/1.54/rnode_firmware_heltec32v2.partitions  
[07:18:14] Firmware extracted
[07:18:14] Flashing RNode firmware to device on /dev/ttyUSB0
esptool.py v3.1
Serial port /dev/ttyUSB0
Connecting....

A fatal error occurred: This chip is ESP32-S3(beta3) not ESP32. Wrong --chip argument?
[ blackops ~ ]#

markqvist, 2022-12-16:
Hmmm, interesting. Either they updated the board design to use an ESP32-S3 (but I doubt that, since they are phasing out the V2 board production), or it is actually another board you have ;)

Is this the board you have?

[image]

killabit, 2022-12-16:
No I have this one..

Heltec ESP32 LoRa 32 (V2), ESP32 Development Board WiFi Bluetooth LoRa Dual Core 240MHz CP2102 with 0.96inch OLED Display Included 868/915MHZ Antenna 

[image]

killabit, 2022-12-16:
I would prefer to use the USB-C as I am going to create a phone case with this board built in with a ultaslim LiPo specifically for use with Sideband for my clients.. so would be great to not have to redesign for an old micro USB board with an adapter..

markqvist, 2022-12-16:
Yeah, that is definitely the newer V3 revision. Was it sold as the V2 board? Would you mind sharing where you bought it from? Just curious.

I've been a bit hesitant to go out and buy a bunch of the V3 boards and add support, since I've seen a lot issues with the V2 boards, and had several reports from other seeing the same.

markqvist, 2022-12-16:
But I get you about the USB-C port, much easier to deal with.

killabit, 2022-12-16:
ooh they are all over Amazon, MakerFocus, HelTech, HiTeach, all V3 Lora32 being sold as V2 and V3  

https://www.amazon.com/MakerFocus-Development-Bluetooth-0-96inch-Display/dp/B076MSLFC9/ref=sr_1_3?crid=S6Q1UCYFTBNR&keywords=heltec+esp32+lora&qid=1671230058&s=electronics&sprefix=heltec+esp32+lora%2Celectronics%2C131&sr=1-3

killabit, 2022-12-16:
esp when everything else is USB-C that piggybacks on this phone, poor android battery gets worked over lol

markqvist, 2022-12-16:
Right, was just curious if it was a common thing for boards ordered as V2 to arrive as V3, but it probably is.

Also, since it's running the SX1262 chip, which is a nice step forward, but is going to depend on this: https://github.com/markqvist/RNode_Firmware/issues/10

killabit, 2022-12-16:
Ahh waiting on hardware? need hardware?

killabit, 2022-12-16:
and just for reference Sideband is getting pre-loaded on all Obsidian Secure Phones as an additional form of encryption communications along with all the other cool stuff it has.. That should give a nice boost to the project.

markqvist, 2022-12-16:
> Ahh waiting on hardware? need hardware?

I *always* need hardware ;)

> Sideband is getting pre-loaded on all Obsidian Secure Phones as an additional form of encryption communications

Interesting. Let's have a chat about that. Are you reachable on Sideband or Matrix?

killabit, 2022-12-16:
I have sideband on TCP on the Sideband default network

markqvist, 2022-12-16:
Can you send an announce or let me know what your LXMF address is?

killabit, 2022-12-16:
yea just sent announce will send again

killabit, 2022-12-16:
2f68def79bfa0ce8cb427bdf255f2e0e

spaceinvadr, 2022-12-27:
Just want to note that I'm running into the same issue as whitecd with firmware version 1.54 on multiple Heltec32v2 devices connected to multiple host machines running Ubuntu 22.04. I'm able to use the device successfully with loramon but when attempting to run rnstatus, rnsd, or nomadnet I get the same errors they mentioned. 

@whitecd were you ever able to get it working on 22.04? I can downgrade the OS if needed but I'd prefer not to if I don't have to.

killabit, 2022-12-31:
> But I get you about the USB-C port, much easier to deal with.

May we please have USB-C this dongle to dongle shit sucks [image]

Dant3s, 2023-02-02:
Hi all :wave:  Have the same issue with Haltec 32v2. Working all good so far when using with Sideband (via BT) with Power Supply (via Phone or via Battery pack). But using via pi3/linux (ubuntu/Fedora) usb interface get the constant port busy or 

/ `A serial port error occurred, the contained exception was: [Errno 9] Bad file descriptor` 

Even if from time to time it works it just flood the console with error logs after that., anyone found a way to make it work?

killabit, 2023-02-10:
> > Ahh waiting on hardware? need hardware?
> 
> 
> 
> I *always* need hardware ;)
> 
> 
> 
> > Sideband is getting pre-loaded on all Obsidian Secure Phones as an additional form of encryption communications
> 
> 
> 
> Interesting. Let's have a chat about that. Are you reachable on Sideband or Matrix?

So now whenever I order V2 hardware i'm being sent v3 hardware, I was thinking of pushing several hundred bluetooth "keychain" rNode devices out there, do you need me to send one so we can get support for it?  Happy to send one either way!  V3 will solve a bunch of portability issues too..

killabit, 2023-03-01:
> Can you send an announce or let me know what your LXMF address is?

Mark, my sideband is down atm, need to chat and figure this out because everyone that is ordering parts to install rNode and build out are being shipped V3 even though specifically requesting V2 from suppliers.  Looks like manufacturing has phased out or sunset the V2 board.

PlanetOfTheGreenPigs, 2023-03-01:
I am also looking into when V3 support will be rolled out. The USB C is nice to have, just need the support sorted for a build and device packaging that I am working on.

markqvist, 2023-03-02:
As soon as the last parts of the reference implementation and API is finished, my work will most likely focus on the SX1262 driver needed for this. I have all the hardware needed to implement the support now as well, so it is "just" a matter of writing the driver, and adding all the configurations to the entire build+delivery+flashing system.

markqvist, 2023-05-02:
Closing this as original issue was resolved

closed 2023-05-02 by markqvist