markqvist/RNode_Firmware #26: Support for LilyGO TTGO-LORA32 V1 868 Mhz board

raytechnical, 2022-11-22:
Hi Mark,

I have RNode working on the TTGO Lora32 Ver 1 board, as the board is not officially supported the OLED display is not working or the LED's, with the exception of a continuous flashing red one when powered up. 

How easy would it to modify the firmware to rectify these minor issues, otherwise everything else works fine, could this be added as feature request please.

markqvist, 2022-11-30:
Hi @raytechnical

Sorry for being slow in getting back to you on this one!

If you can give me the pin definitions you used in your build, and any other necessary information or bits you had to change, I should be able to add official support to the builds and `rnodeconf`. If you can also tell me the info of the OLED, I should be able to get that going too. Can you tell me the following:

- It is also a 128x64 pixel monochrome display, right?
- What i2c address does it use? Probably 0x3C or 0x3D?
- What pins are the i2c clock and data lines on? Is there a physical reset line going to the display, and if so, what pin?

With that info I can throw together a quick test build, and if everything works as expected, we can add it to the release repo :)

raytechnical, 2022-12-25:
Hi Mark,

The OLED display is a mono 0.96 inch, looking at the  port definitions as the diagram below, RESET_OLED pin 16

Additional Meshtatsic pin definitions from their firmware for this board:-

VEXT_ENABLE 21 // active low, powers the OLED display and the Lora antenna boost
LORA_RESET pin 14
LORA_DIO1 pin 35 // Not really used
LORA_DIO2 pin 34
LORA_DIO0 pin 26 // a No connect on the SX1262 module

 I2C_SCL pin 15
I2C_SDA pin 4 // I2C pins for this board

lora32_v1.0_pinmap.jpg

Rnode appears to work with the TTGo version 2 firmware except their is no RX or TX leds or OLED display.

I hope that the above helps.

markqvist, 2022-12-25:
Thanks, it sure does.

If I make a testing build based on the above info, can you try and flash it and report back how it works?

> LORA_DIO0 pin 26 // a No connect on the SX1262 module

Hm, that is strange. Does that mean to say that the DIO_0 from the LoRa transceiver is not connected to the MCU? If that is true, that will bee a problem, since RNode needs to use the DIO_0 pin as an interrupt pin from the transceiver.

> Rnode appears to work with the TTGo version 2 firmware except their is no RX or TX leds or OLED display.

Very interesting. Does both send and receive work? Did you try it out with Reticulum, or only something more passive, like LoRaMon?

raytechnical, 2022-12-25:
Hi Mark,
The pin out details are taken from the Meshtastic distro of their firmware which gives full functionality to this hardware. Have a look at their GitHub page. I am sure that you will be able to double check my findings from their code to trace the pin outs for the SX1262 module to the MCU.

I have only tested the RNode firmware in TNC mode and had it running successfully communicating with another RNode running the version 2 firmware also.

If you can build a version of Rnode then I will give it a beta test using my version 1 TTGo boards.

I need to get around to do some more experimentation with Reticulum, I have only played around using it on my LAN with a Pi and laptop.

raytechnical, 2023-03-06:
Hi Mark,

Just a quick update, did you managed to produce a build for the LilyGO TTGO-LORA32 V1board yet as I am keen to do some more experimentation with it. 

Many thanks.

raytechnical, 2023-03-29:
> Hi Mark,
> 
> Just a quick update, did you managed to produce a build for the LilyGO TTGO-LORA32 V1board yet as I am keen to do some more experimentation with it.
> 
> Many thanks.

Futher

raytechnical, 2023-03-29:
I have carried out some further investigations on the board.
Having modified the TTGO ver 2 and modified the RST pin to GPIO 14 and the LED to pin_led_rx = 25 pin_led_tx = 2. 
I now have a working board which I have tested with Nomadnet and in TNC KISS mode.
The OLED display is not working but I think that this may be down to GPIO pin 21 which controls the power to the display?

erickdkfz, 2023-04-28:
> > Hi Mark,
> > Just a quick update, did you managed to produce a build for the LilyGO TTGO-LORA32 V1board yet as I am keen to do some more experimentation with it.
> > Many thanks.

Would you be willing to share the code for the build you got to work? I tried my hand at it and failed. Would love to compare something that works to whatever nonsense I did!

markqvist, 2023-05-02:
See #35

raytechnical, 2023-05-04:
Hi,
I have uploaded the new V1 firmware and all seems to be working. A big thank you for all your help in getting this to work. 

I will keep be doing some testing over the next few days.

markqvist, 2023-05-04:
Awesome! Please do report back with what you find, I completely rely on the feedback as I don't have that board.

If you have already tested any of the following things, can you confirm that:

- The display is working?
- On-the-air transmission and reception is working?
- Connection to a computer running RNS over USB or serial is working?
- Connection to Android apps over Bluetooth is working?

Thanks for your help!

markqvist, 2023-05-19:
Have you had any time to test this out further? Is everything still operating as expected?

raytechnical, 2023-05-20:
The board is working well at the moment, tested so far:-
1. Display works well and can be disabled as necessary with software.
2. Tested with USB connection with RNS, nomad net and  KISS mode.
3. Haven't had chance to test the Bluetooth connectivity.

markqvist, 2023-05-20:
Awesome, thanks for the feedback! I'll close this issue now as resolved. If any bugs do crop up, let me know :)

closed 2023-05-20 by markqvist