markqvist/RNode_Firmware #29: LoRa TNC Mode Not Receiving

iqw1345, 2023-01-23:
2 x LilyGO LoRa32 v2.1 (aka T3 v1.6.1) running 1.55 firmware
2 x Arch Linux Computers

Hello Mark,

When in TNC mode with both nodes configured identically, the units are unable to communicate with each other. If the antennas are replaced with a direct wire, communication is able to occur and I am able to ping across the LoRa nodes. I can verify that this activity is happening by using tncattach in verbose mode.

Here are the commands being ran:
rnodeconf /dev/ttyACM0 -T --freq 915000000 --bw 500000 --txp 17 --sf 12 --cr 7
sudo tncattach /dev/ttyACM0 115200 --mtu 496 --noipv6 --noup -v

When the exact same configuration is used in reticulum, everything works as intended. I am able to use nomadnet to exchange messages over the LoRa nodes without a problem.

If you need any supporting logs or further documentation, please feel free to let me know.

Thanks in advance.

markqvist, 2023-01-23:
Thank you for reporting this! I've just heard similar from one other user, and I am investigating the issue. Thanks for the additional infor you provided.

One thing I'd like to know is, if reception starts working once a packet has been *sent* in TNC mode? Ie., if you send a ping to some address in the configured IP subnet (which should trigger an ARP request and a packet being sent), does reception start working?

iqw1345, 2023-01-23:
So, here is how I have been testing this (and to provide more context):

I configure both sides with the exact same configuration and then I begin to run ping in flood mode to the other side. At this point, the pings are failing. As ping continues to run (and fail), I remove both antennas and replace it with a wire. At which point, I am able to see pings begin to succeed and the green light begins to flash on the receiving side.

To answer your question:

If you are referring to the reception indicator, yes, after both sides are configured, connected via a jumper wire, and I begin to ping, I get about 2-3 bars on the Q-S meter. When using the antennas, there is no reception.

If you are referring to reception in general, I don't see any indication that packets are being received on the device (when using antenna).

In my humble opinion, it almost seems like the receiver is deaf.

markqvist, 2023-01-24:
Could you try updating the devices to version `1.56` using `rnodeconf`? It should be enough to run:

`rnodeconf --update /dev/ttyACM0`

I believe the issue is fixed in version `1.56`, which I just released, but would like to hear your confirmation.

iqw1345, 2023-01-24:
Works perfectly! Thank you!

markqvist, 2023-01-24:
Awesome! Good to hear :)

Closing this issue as completed then.

closed 2023-01-24 by markqvist