markqvist/RNode_Firmware #31: RGB Led Heltec 32 v2

Dant3s, 2023-02-04:
Hi :wave: 

Im tryng to setup the RGB Led, im using a breakout ws812 and a Heltec 32 v2. 

Cheking the config.h (https://github.com/markqvist/RNode_Firmware/blob/1.57/Config.h) and the guide (step 7 https://unsigned.io/guides/2023_01_14_Making_A_Handheld_RNode.html ) I see that the io pin for the data in the board is other than the IO12 Right?  25 or 36/37. 

Im getting that right? 

Reference: https://escapequotes.net/wp-content/uploads/2021/02/WIFI_LoRa_32_V2-Heltec-pinout-diagram-scaled.jpg

markqvist, 2023-02-06:
I can't tell from your post, did you enable the RGB LED (HAS_NEOPIXEL) for the Heltec board in Config.h?

Dant3s, 2023-02-06:
Ah! looks like that is what is missing. Didn't notice that by default HAS_NP in false? and the Heltec entry don't have that  if  HAS_NP (I would assume NP is neopixel?) 

```
			#if defined(EXTERNAL_LEDS)
				const int pin_led_rx = 36;
				const int pin_led_tx = 37;
			#else
				const int pin_led_rx = 25;
				const int pin_led_tx = 25;

```

Would assume that should be inside an if, right? : 

`			#if HAS_NP == false
`

This will require to compile the kernet into the board ? There is no config with **rnconfz**?

markqvist, 2023-02-06:
That's correct, `HAS_NP` is the config directive for enabling the RGB LED. It's off by default since having it on disables the other LEDs.

I could add a Heltec build with RGB LEDs enabled if there's interest, but right now you'd have to compile it yourself with those changes.

closed 2023-02-07 by Dant3s