markqvist/RNode_Firmware #32: Heltec v2 Battery Support

Dant3s, 2023-02-13:
Hi! :wave: 

I notice that the fw has by default the battery support disable. I would assume that as in the RGB Led part, this need to be setup at FW and compile. Right? Does this options will always be Compile and upload or would at some point be part of the rnodeconf? 

Also.. How is the support for Battery Charge/Discharge for Heltec v2? Any thing that need to be done, besides defining the battery flags and values at the conf.h? 

Is possible to understand the use for:

    bool battery_installed = false;
    **bool battery_indeterminate = false;
    bool external_power = false;
    bool battery_ready = false;**
    float battery_voltage = 0.0;
    float battery_percent = 0.0;

Thanks!

markqvist, 2023-02-13:
The battery charge and discharge will already work, but the firmware will just not have any idea about the battery percentage or charge state. So you can use a battery just fine.

Do you know if the board has the battery voltage connected to one of the ADC pins?

Dant3s, 2023-02-13:
Acording to the Layout (https://resource.heltec.cn/download/WiFi_LoRa_32/WIFI_LoRa_32_V2.pdf)  and issues in meshtastic, it does: 

https://github.com/meshtastic/firmware/issues/442
https://meshtastic.discourse.group/t/heltec-lora-32-v2-1-battery-voltage-reading/5678/3

If  it looks like too much of a hassle, could be something for the future and will stick with my power bank idea for simplicity.

markqvist, 2023-02-13:
I'll add it at some point :) Should be a pretty minor addition to make.

closed 2024-02-29 by markqvist