markqvist/RNode_Firmware #33: windows 10, rnodeconf --autoinstall, " You do not currently have the "unzip" program installed on your system"

pizzadoglover911, 2023-03-05:
I am a novice, so please bear with me. 

Of course I am missing "unzip" program, but what is the "unzip" program specifically? The command suggested after the error is only relevant on Linux(sudo apt install unzip). 

Is there any way to utilize the config on Windows or does this error suggest the neccesity to use a Linux distribution and "unzip" package?

Thanks

markqvist, 2023-03-05:
Hi @pizzadoglover911! The autoinstaller is not officially supported on Windows, but other users have reported success by using it in WSL.

I'll close this as an issue, but you are of course welcome to ask any additional questions here.

closed 2023-03-05 by markqvist

pizzadoglover911, 2023-03-05:
Thanks for the quick reply! I kept running into an issue on my ubuntu laptop. Traceback excecptions for Cryptography, so I gave up and tried on windows. 

A fresh install of ubuntu resolved that Cryptography execption and I have one of my Heltec boards displaying RNODE, v1.5, harware init okay, trx ready. That auto installer is pretty slick. Now to figure out how to use all this. 

is there any method of donation other than the coins you have listed?

pizzadoglover911, 2023-03-05:
> Thanks for the quick reply! I kept running into an issue on my ubuntu laptop. Traceback excecptions for Cryptography, so I gave up and tried on windows.
>
> A fresh install of ubuntu resolved that [...]

found it, cancel that