markqvist/RNode_Firmware #34: Unable to upload from Makefile followed by corrupt firmware

acehoss, 2023-03-11:
I was trying to build from source and use the upload target in the Makefile. I have an Unsigned RNode gen 2 900 MHz. It was working fine and running RNode v1.57.

1. `make firmware-rnode_ng_21`
2. Edit makefile to use serial port `/dev/cu.wchusbserial54F70179081`
3. `make upload-rnode_ng_21`

The upload works, but the firmware does not start. I unplugged and replugged, the screen just stays black. So then I used `rnodeconf -a` to restore the firmware, and now the screen says `FIRMWARE CORRUPT` and the LED is slow blinking red.

[image]

acehoss, 2023-03-11:
I did get `make upload-rnode_ng_21` to work from a Linux box, and the corrupt firmware issue went away. Probably we can close this for now because someone would need to be digging deep to get themselves into this problem.

closed 2023-03-11 by acehoss

markqvist, 2023-03-11:
Hmm... That is odd. There might be something not working as expected on macOS when setting the firmware hashes after flashing then. That would cause that error.