markqvist/RNode_Firmware #46: Support for Heltec Wireless Stick (Gecko Board) - LoRa node chip SX1276/SX1278 - Screen issue

hhglae, 2023-10-03:
I tried out today the RNode Firmware on a Heltec Wireless Stick. It seem to boot in general but the screen is not correct. The reason seems to be correct, because i installed the only available image for heltec, the one for "Heltec LoRa32 v2" in version 1.64. The Heltec Wireless Stick have a completely different screen size and screen type. I used rnodeconf --autoinstall to install it.

The one i have is: https://heltec.org/project/wireless-stick/

I think the screen issue should be a simple fix for a developer.

markqvist, 2023-10-03:
Interesting, I wouldn't have expected that to work.

If the firmware works on the board, that's great! And also a bit of a stroke of luck, since the pin mappings would need to match the supported board.

EIther way, that board uses a 64x32 pixel display. All other boards with a supported display use 128x64 resolution, so there's not really a "simple fix" for anyone to just make here. It would require a completely new display driver, new graphics, new layout and would only be able to display a quarter of the information of the supported displays.

Unfortunately, I don't think anyone is going to take up the challenge for now, but if you are up to it, and feel it may be an easy job, you're very welcome to work on it and submit a pull request supporting this particular board and screen!

closed 2023-10-03 by markqvist

hhglae, 2023-10-03:
> It would require a completely new display driver

No, its outputting information on the screen. Its 90° turned to the side and because it have less pixel, the menu is cut.
I was hoping that this would be some scaling-thing that have to be enabled to be at 50% and the screen size or pixel have to be configured and that would be all required.

markqvist, 2023-10-03:
Ah, that's really interesting! I would really not have expected it to output anything at all! Apparently, that display must be using the same driver chip and I2C address then.

I get you. Unfortunately it won't be that easy. The 128x64 displays are already using more or less every pixel available, with tiny fonts in most cases. We just can't scale it down any more - then it will be completely unreadable ;)

One could create a "micro-screen" version, but it's a pretty big effort to redo everything, so I doubt it will happen in the near future.

hhglae, 2023-11-05:
[image]
[image]

The solution is simple. Connect a screen with the correct amount of pixels. Because its a i2c-bus both screens work at the same time. To save energy you should disconnect the smaller one.