markqvist/RNode_Firmware #48: LILYGO T-Echo support

nowozin, 2023-11-06:
The [T-Echo](https://www.lilygo.cc/products/t-echo) device from LILYGO is very similar to the T-Beam devices.

Additional information:
* The T-Echo uses the SX1262 chip.
* _Meshtastic_ supports T-Beam and T-Echo and may be a useful source in case there are subtle differences ([hardware devices](https://meshtastic.org/docs/supported-hardware), [development information](https://meshtastic.org/docs/development/firmware/build))

I am happy to fund the purchase of a T-Echo device (around 80 USD for device + shipping fee) in case a developer with suitable expertise is willing to port the RNode firmware to the T-Echo devices.

nowozin, 2023-11-10:
Ping on this.

faragher, 2023-11-10:
Mark is currently in an area with limited connectivity, and he's the one to ask about the firmware. It will likely take more than a few days, possibly into the new year, before he can do any modifications or feasibility studies.

nowozin, 2023-11-12:
Thanks for the information @faragher. I will be patient then :+1:

markqvist, 2024-02-29:
No takers for implementation. Closing as out of scope for now.

closed 2024-02-29 by markqvist