markqvist/RNode_Firmware #49: G-NiceRF Lora1278F30

ahedproductions, 2023-11-13:
Hello!
I am waiting on two items from AliExpress:
1) Two piesec  ESP-32 38Pin board from this posting: https://www.aliexpress.com/item/1005004230550198.html
2) Two pieces G-NiceRF Lora1278F30 module with 30dBm max power from this post.
These will allow me to experiment ang get two functional RNodes, for which I have prepared a fresh install on a RaspberryPI4 B of Raspbian + Reticulum. If I succeed in setting those together, I will create a PCB, which I will fully share with the community (all design files), which will house an ESP32 and the LoRa modem for a compact unit, capable of 1W output power for long-distance comms, on top ow which I plan to use some kind of minimal raspberrypi setup to run Reticulum and create something standalone similar to Meshenger at first as some kind of addon module. Then I will do a pure Transport node to be used as LoRa repeater, which will be a different addon board, then a internetworking bridge, to be able to get long-haul over, MMWave, Wi-Fi, HF, VHF etc.
The only things I am not sure of, is if I will be able to use autoconfiguration for this module and how to set output power from rnodeconf, all those modem settings look to me only apply to TNC mode?
Attached is all I have for the two modules, looks like the AZ one:
[ESP_-_32_NodeMCU_Developmentboard_Datenblatt_AZ-Delivery_Vertriebs_GmbH.pdf](https://github.com/markqvist/RNode_Firmware/files/13331324/ESP_-_32_NodeMCU_Developmentboard_Datenblatt_AZ-Delivery_Vertriebs_GmbH.pdf)
[ESP-32_NodeMCU_Developmentboard_Pinout.pdf](https://github.com/markqvist/RNode_Firmware/files/13331325/ESP-32_NodeMCU_Developmentboard_Pinout.pdf)
[ESP-32_NodeMCU_Developmentboard_Schematic_korr.pdf](https://github.com/markqvist/RNode_Firmware/files/13331327/ESP-32_NodeMCU_Developmentboard_Schematic_korr.pdf)
[LoRa1278F30 1W Wireless Tranceiver Module V3.1.pdf](https://github.com/markqvist/RNode_Firmware/files/13331349/LoRa1278F30.1W.Wireless.Tranceiver.Module.V3.1.pdf)
[image]
I will appreciate any help, I am not a software guy, but a hardware, RF designer.
All the best!
Bill
LZ1SWE

markqvist, 2023-11-13:
That's a great project. A higher-powered RNode is something I know quite a few people have been interested in!

Since the LoRa module provides direct access to the SPI bus and DIO interrupt pins, it looks like it should work just fine with the firmware as it is.

The only detail is the higher output power, and that is something we will need to add support for in the firmware, by creating a new board definition, with the relevant "output dBm to configuration value mappings", and also add this to `rnodeconf`. Don't worry, I can take care of this part easily.

Once you get the parts in, and get a basic setup working, let me know, and we'll get the new board definition in there. If you can think of a name/specifier to use for your RNode design in `rnodeconf`, that would be great too. Then we can give it it's own entry for easy installation with autoinstall and such.

ahedproductions, 2023-11-13:
Thank you for the response and support! 
Well, my lab is called aetherlab, aethernode should fit well.
For the development phase I will use the stock antennas and low power, maybe the 5-20mW range, but the final version should be able to either be power tunable by the user, or maybe use RSSI and auto-adjust, we will see what is best. I have already started working on a new antenna design for the center of the 430 ISM band, which will be a lot better than the stock.
I sent you several messages to your email the last several days, were you able to see those? From aetherlab at tutanota dot kom?

markqvist, 2023-11-13:
Yes, I got them, and will respond to you, but I am currently outside of constant internet access, and only have intermittent connectivity and ability to answer.

ahedproductions, 2023-11-17:
Hello! A quick update on the project. While waiting for parts for the _aethernode_ I am working on a new 433MHz base antenna for repeater nodes, which will later be scalable to higher frequencies. I have based my design on the proven work of MP Antenna, and while it is not a simple design, it will provide for increased coverage in non-line-of-sight (NLOS) environments by having a multi-polarization pattern. This will allow for better coverage, especially with low-power RNodes, which we have to go for to get highly mobile and user-friendly Reticulum-based standalone devices (which do not rely on any other "modern" tech, which is highly security compromised on the hardware level, as chipsets have hardware-based built-in backdoors for surveillance and control). 
The original MP 08-ANT-0891 looks like this:
[image]
The multi-polarization pattern is secured via splitting the vertical part into three rods of slightly different length and bending down the ground-plane rods, in this case 8 pieces, instead of the traditional 3 rods. The antenna has a very complex metal centre part for the verticals and I am working on redoing and a bit simplifying this design, while retaining the multi-polarization pattern as much as possible, while making the antenna easy to reproduce for anyone with relatively basic means. This will be a long process, but I will start on the nodes as soon as they arrive in the mail and together with Mark get them up and defined in confs.
I made a logo for the aethernode
[image]

ahedproductions, 2023-12-20:
Hello!
Shipping has been a complete nightmare! I received the LoRa modules today, still waiting for the ESP32's.
[image]
I have created an Altium Library for it, which you can find attached.
[1278F30 Altium Designer Libraries.zip](https://github.com/markqvist/RNode_Firmware/files/13729992/1278F30.Altium.Designer.Libraries.zip)
I hope the ESP32's arrive soon, so I will have the holidays to get them up and running.
All the best to everyone and happy holidays!
Bill

ahedproductions, 2024-01-02:
Happy new year! I wish everyone health, love and peace in 2024.
It has been a while, but they are finally here!
[image]
The moment I get around to getting them paired with the LoRa modules, I will post, Mark. I hope we will be able to get them running quickly.
All the best!
Bill
LZ1SWE

ahedproductions, 2024-01-08:
[image]

Boards are ready to roll! I also have a Raspi 3b PC with all installs ready. I am on standby!
All the best!
Bill
LZ1SWE

markqvist, 2024-01-16:
Hey Bill
That is looking great man. Here's a `Config.h` file with product config added for Aethernode. You can use those values as a starting point for adding support to the firmware. At some point you will also need to

- Add model codes in `ROM.h`, for the different models you plan to make.
- Specify LoRa module config regarding to RF output pin in `Utilities.h` line 992.

But for now, I'd suggest just using the already included model codes for generic ESP32 boards. This way you can also just initially flash and bootstrap your boards via the `rnodeconf` tool, and when that has set up the EEPROM config you can just flash your own firmware builds via the makefile from your local firmware source directory.

You will probably also want to add specific sections for Aethernode to the makefile. You can base those off any of the other boards that you'd want.

```cpp
// Copyright (C) 2023, Mark Qvist

// This program is free software: you can redistribute it and/or modify
// it under the terms of the GNU General Public License as published by
// the Free Software Foundation, either version 3 of the License, or
// (at your option) any later version.

// This program is distributed in the hope that it will be useful,
// but WITHOUT ANY WARRANTY; without even the implied warranty of
// MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
// GNU General Public License for more details.

// You should have received a copy of the GNU General Public License
// along with this program.  If not, see <https://www.gnu.org/licenses/>.

#include "ROM.h"

#ifndef CONFIG_H
	#define CONFIG_H

	#define MAJ_VERS  0x01
	#define MIN_VERS  0x44

	#define PLATFORM_AVR   0x90
  #define PLATFORM_ESP32 0x80

	#define MCU_1284P 0x91
	#define MCU_2560  0x92
	#define MCU_ESP32 0x81

	#define BOARD_RNODE         0x31
	#define BOARD_HMBRW         0x32
	#define BOARD_TBEAM         0x33
	#define BOARD_HUZZAH32      0x34
	#define BOARD_GENERIC_ESP32 0x35
	#define BOARD_LORA32_V2_0   0x36
	#define BOARD_LORA32_V2_1   0x37
	#define BOARD_LORA32_V1_0   0x39
	#define BOARD_HELTEC32_V2   0x38
	#define BOARD_RNODE_NG_20   0x40
	#define BOARD_RNODE_NG_21   0x41
  #define BOARD_AETHERNODE    0x80

	#define MODE_HOST 0x11
	#define MODE_TNC  0x12

	#define CABLE_STATE_DISCONNECTED 0x00
	#define CABLE_STATE_CONNECTED    0x01
	uint8_t cable_state = CABLE_STATE_DISCONNECTED;
	
	#define BT_STATE_NA        0xff
	#define BT_STATE_OFF       0x00
	#define BT_STATE_ON        0x01
	#define BT_STATE_PAIRING   0x02
	#define BT_STATE_CONNECTED 0x03
	uint8_t bt_state = BT_STATE_NA;
	uint32_t bt_ssp_pin = 0;
	bool bt_ready = false;
	bool bt_enabled = false;
[... 349 lines cut ...]
    #define BATTERY_STATE_DISCHARGING 0x01
    #define BATTERY_STATE_CHARGING 0x02
    #define BATTERY_STATE_CHARGED 0x03
    bool battery_installed = false;
    bool battery_indeterminate = false;
    bool external_power = false;
    bool battery_ready = false;
    float battery_voltage = 0.0;
    float battery_percent = 0.0;
    uint8_t battery_state = 0x00;
    uint8_t display_intensity = 0xFF;
    uint8_t display_addr = 0xFF;
    bool display_diagnostics = true;    
    bool device_init_done = false;
    bool eeprom_ok = false;
    bool firmware_update_mode = false;

	// Boot flags
	#define START_FROM_BOOTLOADER 0x01
	#define START_FROM_POWERON 0x02
	#define START_FROM_BROWNOUT 0x03
	#define START_FROM_JTAG 0x04

#endif
```

markqvist, 2024-01-16:
By the way, the antenna project is amazing. It'd be a big help to have  something like that available relatively cheaply, and I would definitely want one myself ;)

ahedproductions, 2024-01-16:
Hello! Glad you are back!
Wow, you have a lot of faith in me! It took me 4 hours to successfully setup Reticulum on the Pi, which is a 4 by the way, not 3B. I am not a Linux guy at all! Most of the time I spent reading about CLI commands... I will do my best, following the instructions from the RNode page. And I hope I find the location of the above mentioned files easily.
About the antenna - I gave up on it. After almost 2 weeks of modifying the model in 4nec2, I finally found out what exactly they have designed. While it would most certainly provide what was promised, it is VERY finicky and if one does not follow it to the mm (or less), the results are very much worse. It has a very nice multi-polar diagram, but I tried with just a slight ice buildup, and it was almost useless. For this to be at it's best, it needs some kind of a radome, which defies all meaning, if we focus on low price and ease of repeatability, at least for the prepared. Anyway, by the time I get into the 5-10-30 watts for the repeaters, simple high quality vertical sleeves (https://www.digikey.bg/en/products/detail/taoglas-limited/OMB-433-B06F21/6362795?s=N4IgjCBcoCwwrFUBjKAzAhgGwM4FMAaEAeygG0QAmANmvgAYBmEAXSIAcAXKEAZU4BOASwB2AcxABfImAAc8CNBCpImXIRLkqcgOwBOZkWqM49GCCKU5s%2BhEvX6ie7Js7WHbpD6DRE6eD1ZPSRldGx8IlJICnpWfwBaYKUVQQBXDSiKCBYEyhCUgXTIrUQc-2oQoQATHniwWwsQLh5GzgBPdjweKpxUSUkgA) and cheaper 1/4 waves for the static setups (https://www.digikey.bg/en/products/detail/siretta-ltd/TANGO9-1-5M-SMAM-S-S-20/12179148), will do!
I will do my best to get the boards working and get back to you soon!
All the best!
Bill
73 de LZ1SWE

markqvist, 2024-01-16:
Ah, okay, nice effort on it anyways. Would have been very nice to have something with those parameters, but I agree that's way too much effort for the average guy. Sounds like a good plan with simple high-quality verticals. Have had very good success with similar myself.

You'll get it up and running no problem :) You obviously need to adjust the pins too, but it should all be more or less logical to figure it out, once you get into it. Otherwise, just modify the already included "Generic ESP32" configuration, and we can look at the diffs between your repo and the master later, and separate that out into a specific configuration for your board.

Just install the "official" firmware with `rnodeconf` first, so it gets the correct EEPROM config and such. You can then compile and install your own modifications from the source.

faragher, 2024-01-16:
Thanks for the heads up. Around here something that can't handle ice gets pretty useless up to half the year.

ahedproductions, 2024-01-21:
Well, darn! Both boards give back - A fatal error occured: Failed to connect to ESP32: Timed out waiting for packet header.
I tried even at 9600. The LED blinks at tries, voltages (5 and 3.3) are stable on the board. Should I disconnect the RF module for the programming part? And, also, it asks me if the antenna on the RF module is at RF0 or BOOST pin?

ahedproductions, 2024-01-21:
Sorry for the dumb question everyone, but:
I have only one config.h on my system and it, as I can see it's contents, is not related to Reticulum at all. Where should I place the new config.h, in what directory? And where can I find the others mentioned above, ROM.h and utilities.h?

jacobeva, 2024-01-23:
You need to add those pins with a new board definition into the config.h in the RNode_Firmware repo cloned on your computer. ROM.h and Utilities.h should be in the same directory, make sure you've changed directory into the RNode_Firmware repo on your filesystem. Hope that helps! For examples of board definitions, look at the existing ones in Config.h, e.g. https://github.com/markqvist/RNode_Firmware/blob/master/Config.h#L304

ahedproductions, 2024-01-23:
Hello and tanks for the response!
I can prepare the file, no problem, already did, but I cannot find it on the Raspberry. Linux's filesystem is kind of murky to me, where should the repo reside?

jacobeva, 2024-01-23:
The repo will be wherever you clone it to on the Raspberry pi.
You can clone it with the following command: `git clone https://github.com/markqvist/RNode_Firmware`

ahedproductions, 2024-01-24:
I see! I did not do this. I installed Reticulum only. And as rnodeconf is there somewhere, it just ran from terminal, and somehow told me, that the process was complete, I think it pulled the default firmware online.
I had no idea. OK, I will and try again, thanks for the feedback!

ahedproductions, 2024-01-27:
Allright!
I have cloned RNode_Firmware. I have changed the config.h file to the one Mark gave me. I have set the pins correctly in config.h.
And now when I run rnodeconf where and how must I select AETHERNODE? I does not appear anywhere?

ahedproductions, 2024-02-03:
Help, anyone?

markqvist, 2024-02-08:
I'll be with you as soon as I can Bill, just a bit overloaded right now

ahedproductions, 2024-02-08:
Oh sure! Thanks, just wondering if anyone was reading the thread! I am on standby!

markqvist, 2024-02-08:
I was unable to get to a computer for a while, and had some rather pressing concerns elsewhere ;)

ahedproductions, 2024-02-08:
I truly hope everything with you is fine! :)

jacobeva, 2024-02-10:
> And now when I run rnodeconf where and how must I select AETHERNODE? I does not appear anywhere? Maybe more dumb questions on the way! :)
You need to flash the board using the Makefile, take a look at the Makefile and see the commands that are run for each target. In this case, you probably want to compile for the generic esp32 target, so `make firmware-genericesp32 && make upload-genericesp32`.

jacobeva, 2024-02-10:
Run that from the RNode_Firmware dir of course.

ahedproductions, 2024-02-11:
root@reticulum:/home/reticulum/RNode_Firmware# make firmware-genericesp32 && make upload-genericesp32
arduino-cli compile --fqbn esp32:esp32:esp32 -e --build-property "build.partitions=no_ota" --build-property "upload.maximum_size=2097152" --build-property "compiler.cpp.extra_flags=\"-DBOARD_MODEL=0x35\" \"-DMODEM=0x01\""
/bin/sh: 1: arduino-cli: not found
make: *** [Makefile:103: firmware-genericesp32] Error 127
root@reticulum:/home/reticulum/RNode_Firmware#

ahedproductions, 2024-02-11:
Holy *****!! Linux is a tough place for me! I installed arduino-cli via TAR.GZ download and copied it to /bin. Buuut:
Error during build: Platform 'esp32:esp32' not found: platform not installed
Try running `arduino-cli core install esp32:esp32`
make: *** [Makefile:103: firmware-genericesp32] Error 1

ahedproductions, 2024-02-11:
OH COME OOOON!!!
root@reticulum:/home/reticulum/RNode_Firmware# make firmware-genericesp32 && make upload-genericesp32
arduino-cli compile --fqbn esp32:esp32:esp32 -e --build-property "build.partitions=no_ota" --build-property "upload.maximum_size=2097152" --build-property "compiler.cpp.extra_flags=\"-DBOARD_MODEL=0x35\" \"-DMODEM=0x01\""
In file included from /home/reticulum/RNode_Firmware/Utilities.h:44,
                 from /home/reticulum/RNode_Firmware/RNode_Firmware.ino:18:
/home/reticulum/RNode_Firmware/Device.h:16:10: fatal error: Ed25519.h: No such file or directory
 #include <Ed25519.h>
          ^~~~~~~~~~~
compilation terminated.

Used library    Version Path
SPI             2.0.0   /root/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/SPI
BluetoothSerial 2.0.0   /root/.arduino15/packages/esp32/hardware/esp32/2.0.11/libraries/BluetoothSerial

Used platform Version Path
esp32:esp32   2.0.11  /root/.arduino15/packages/esp32/hardware/esp32/2.0.11
Error during build: exit status 1
make: *** [Makefile:103: firmware-genericesp32] Error 1

jacobeva, 2024-02-11:
Please run `make prep-esp32` before trying to run `make firmware-genericesp32`, my bad.

ahedproductions, 2024-02-11:
root@reticulum:/home/reticulum/RNode_Firmware# make firmware-genericesp32 && make upload-genericesp32
arduino-cli compile --fqbn esp32:esp32:esp32 -e --build-property "build.partitions=no_ota" --build-property "upload.maximum_size=2097152" --build-property "compiler.cpp.extra_flags=\"-DBOARD_MODEL=0x35\" \"-DMODEM=0x01\""
In file included from /home/reticulum/RNode_Firmware/Utilities.h:36,
                 from /home/reticulum/RNode_Firmware/RNode_Firmware.ino:18:
/home/reticulum/RNode_Firmware/Bluetooth.h: In function 'bool bt_setup_hw()':
/home/reticulum/RNode_Firmware/Bluetooth.h:92:11: error: 'EEPROM' was not declared in this scope
       if (EEPROM.read(eeprom_addr(ADDR_CONF_BT)) == BT_ENABLE_BYTE) {
           ^~~~~~
/home/reticulum/RNode_Firmware/Bluetooth.h:92:11: note: suggested alternative: 'EIDRM'
       if (EEPROM.read(eeprom_addr(ADDR_CONF_BT)) == BT_ENABLE_BYTE) {
           ^~~~~~
           EIDRM
/home/reticulum/RNode_Firmware/Bluetooth.h:105:37: error: 'EEPROM' was not declared in this scope
             data[BT_DEV_ADDR_LEN] = EEPROM.read(eeprom_addr(ADDR_SIGNATURE));
                                     ^~~~~~
/home/reticulum/RNode_Firmware/Bluetooth.h:105:37: note: suggested alternative: 'EIDRM'
             data[BT_DEV_ADDR_LEN] = EEPROM.read(eeprom_addr(ADDR_SIGNATURE));
                                     ^~~~~~
                                     EIDRM
In file included from /home/reticulum/RNode_Firmware/Utilities.h:44,
                 from /home/reticulum/RNode_Firmware/RNode_Firmware.ino:18:
/home/reticulum/RNode_Firmware/Device.h: In function 'bool device_init()':
/home/reticulum/RNode_Firmware/Device.h:148:70: error: 'EEPROM' was not declared in this scope
     for (uint8_t i=0; i<EEPROM_SIG_LEN; i++){dev_eeprom_signature[i]=EEPROM.read(eeprom_addr(ADDR_SIGNATURE+i));}
                                                                      ^~~~~~
/home/reticulum/RNode_Firmware/Device.h:148:70: note: suggested alternative: 'EIDRM'
     for (uint8_t i=0; i<EEPROM_SIG_LEN; i++){dev_eeprom_signature[i]=EEPROM.read(eeprom_addr(ADDR_SIGNATURE+i));}
                                                                      ^~~~~~
                                                                      EIDRM
In file included from /home/reticulum/RNode_Firmware/RNode_Firmware.ino:18:
/home/reticulum/RNode_Firmware/Utilities.h: In function 'bool eeprom_info_locked()':
/home/reticulum/RNode_Firmware/Utilities.h:1110:6: error: 'lock_byte' was not declared in this scope
  if (lock_byte == INFO_LOCK_BYTE) {
      ^~~~~~~~~
/home/reticulum/RNode_Firmware/Utilities.h:1110:6: note: suggested alternative: 'clock_t'
  if (lock_byte == INFO_LOCK_BYTE) {
      ^~~~~~~~~
      clock_t
/home/reticulum/RNode_Firmware/Utilities.h: In function 'void eeprom_dump_info()':
/home/reticulum/RNode_Firmware/Utilities.h:1124:28: error: expected primary-expression before ')' token
   escaped_serial_write(byte);
                            ^
/home/reticulum/RNode_Firmware/Utilities.h: In function 'void eeprom_dump_config()':
/home/reticulum/RNode_Firmware/Utilities.h:1135:28: error: expected primary-expression before ')' token
   escaped_serial_write(byte);
                            ^
/home/reticulum/RNode_Firmware/Utilities.h: In function 'void eeprom_dump_all()':
/home/reticulum/RNode_Firmware/Utilities.h:1146:28: error: expected primary-expression before ')' token
   escaped_serial_write(byte);
                            ^
/home/reticulum/RNode_Firmware/Utilities.h: In function 'void eeprom_update(int, uint8_t)':
/home/reticulum/RNode_Firmware/Utilities.h:1161:7: error: 'EEPROM' was not declared in this scope
   if (EEPROM.read(mapped_addr) != byte) {
       ^~~~~~
/home/reticulum/RNode_Firmware/Utilities.h:1161:7: note: suggested alternative: 'EIDRM'
   if (EEPROM.read(mapped_addr) != byte) {
       ^~~~~~
       EIDRM
/home/reticulum/RNode_Firmware/Utilities.h: At global scope:
/home/reticulum/RNode_Firmware/Utilities.h:1207:4: error: expected unqualified-id before 'else'
  } else {
    ^~~~
/home/reticulum/RNode_Firmware/Utilities.h:1210:1: error: expected declaration before '}' token
 }
 ^

Used library    Version Path
SPI             2.0.0   /root/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/SPI
BluetoothSerial 2.0.0   /root/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/BluetoothSerial
Crypto          0.4.0   /root/Arduino/libraries/Crypto
FS              2.0.0   /root/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/FS
SPIFFS          2.0.0   /root/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/SPIFFS
WiFi            2.0.0   /root/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/WiFi
WebServer       2.0.0   /root/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/WebServer
SD              2.0.0   /root/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/SD

Used platform Version Path
esp32:esp32   2.0.14  /root/.arduino15/packages/esp32/hardware/esp32/2.0.14
Error during build: exit status 1
make: *** [Makefile:103: firmware-genericesp32] Error 1
root@reticulum:/home/reticulum/RNode_Firmware# make firmware-aethernode && make upload-aethernode
make: *** No rule to make target 'firmware-aethernode'.  Stop.

jacobeva, 2024-02-11:
Run it git pull again

ahedproductions, 2024-02-12:
> Run it git pull again

Sorry, what? Please, please keep in mind - programming/linux environment terminology evades me... :)

faragher, 2024-02-12:
Git is a version control system. If you pulled the repository using 

```git clone https://github.com/markqvist/RNode_Firmware.git```

for example, then it downloaded the current main branch to your machine. It means if you're authorized, you could upload your changes as well, but it helps keep up to date as well. If you go to the main repository directory you can use

```git pull```

to get the current version, overwriting any changes you've made. 

Regarding dependencies, that's a huge kettle of fish that systems like pip or nuget (or Node or whatever for other languages) was meant to correct. Using Linux, the best way, if it's possible, is to look in your distro for the files you need. Often when I do a search for my missing repos online, I find I can install it using apt/yum or whatever your distro uses.

I have no real specific advice to provide about the firmware.

ahedproductions, 2024-02-12:
> Git is a version control system...

Thank you! Such is the instruction I really need! So my understanding of the existence of Git Hub is now complete... :D
I will execute and than edit the Config again, to be able to rebuild correctly. Or a least I will try!
Cheers!

ahedproductions, 2024-02-19:
I did a pull. Now my Config.h file i very much shorter, no boards pin configuration, etc. What the hell is going on?
// Copyright (C) 2023, Mark Qvist

// This program is free software: you can redistribute it and/or modify
// it under the terms of the GNU General Public License as published by
// the Free Software Foundation, either version 3 of the License, or
// (at your option) any later version.

// This program is distributed in the hope that it will be useful,
// but WITHOUT ANY WARRANTY; without even the implied warranty of
// MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
// GNU General Public License for more details.

// You should have received a copy of the GNU General Public License
// along with this program.  If not, see <https://www.gnu.org/licenses/>.

#include "ROM.h"
#include "Boards.h"

#ifndef CONFIG_H
	#define CONFIG_H

	#define MAJ_VERS  0x01
	#define MIN_VERS  0x46

	#define MODE_HOST 0x11
	#define MODE_TNC  0x12

	#define CABLE_STATE_DISCONNECTED 0x00
	#define CABLE_STATE_CONNECTED    0x01
	uint8_t cable_state = CABLE_STATE_DISCONNECTED;
	
	#define BT_STATE_NA        0xff
	#define BT_STATE_OFF       0x00
	#define BT_STATE_ON        0x01
	#define BT_STATE_PAIRING   0x02
	#define BT_STATE_CONNECTED 0x03
	uint8_t bt_state = BT_STATE_NA;
	uint32_t bt_ssp_pin = 0;
	bool bt_ready = false;
	bool bt_enabled = false;
	bool bt_allow_pairing = false;

	#define M_FRQ_S 27388122
	#define M_FRQ_R 27388061
	bool console_active = false;
	bool modem_installed = false;

	#define MTU   	   508
	#define SINGLE_MTU 255
	#define HEADER_L   1
	#define MIN_L	   1
	#define CMD_L      64

    bool mw_radio_online = false;

	#define eeprom_addr(a) (a+EEPROM_OFFSET)

    #if (MODEM == SX1262 || MODEM == SX1280) && defined(NRF52840_XXAA)
        SPIClass spiModem(NRF_SPIM2, pin_miso, pin_sclk, pin_mosi);
    #endif

	// MCU independent configuration parameters
	const long serial_baudrate  = 115200;

	// SX1276 RSSI offset to get dBm value from
	// packet RSSI register
	const int  rssi_offset = 157;

	// Default LoRa settings
	const int lora_rx_turnaround_ms = 66;
	const int lora_post_tx_yield_slots = 6;
	uint32_t post_tx_yield_timeout = 0;
	#define LORA_PREAMBLE_SYMBOLS_HW  4
	#define LORA_PREAMBLE_SYMBOLS_MIN 18
	#define LORA_PREAMBLE_TARGET_MS   15
	#define LORA_CAD_SYMBOLS 3
	int csma_slot_ms = 50;
	float csma_p_min = 0.1;
	float csma_p_max = 0.8;
	uint8_t csma_p = 0;

	int  lora_sf   	           = 0;
	int  lora_cr               = 5;
	int  lora_txp              = 0xFF;
	uint32_t lora_bw           = 0;
	uint32_t lora_freq         = 0;
	uint32_t lora_bitrate      = 0;
	long lora_preamble_symbols = 6;
	float lora_symbol_time_ms  = 0.0;
	float lora_symbol_rate     = 0.0;
	float lora_us_per_byte     = 0.0;

	// Operational variables
	bool radio_locked  = true;
	bool radio_online  = false;
	bool community_fw  = true;
	bool hw_ready      = false;
	bool radio_error   = false;
	bool disp_ready    = false;
	bool pmu_ready     = false;
	bool promisc       = false;
	bool implicit      = false;
	uint8_t implicit_l = 0;

	uint8_t op_mode   = MODE_HOST;
	uint8_t model     = 0x00;
	uint8_t hwrev     = 0x00;

    int     current_rssi    = -292;
	int		last_rssi		= -292;
	uint8_t last_rssi_raw   = 0x00;
	uint8_t last_snr_raw	= 0x80;
	uint8_t seq				= 0xFF;
	uint16_t read_len		= 0;

	// Incoming packet buffer
	uint8_t pbuf[MTU];

	// KISS command buffer
	uint8_t cmdbuf[CMD_L];

	// LoRa transmit buffer
	uint8_t tbuf[MTU];

	uint32_t stat_rx		= 0;
	uint32_t stat_tx		= 0;

	#define STATUS_INTERVAL_MS 3
	#if MCU_VARIANT == MCU_ESP32 || MCU_VARIANT == MCU_NRF52
	  #define DCD_SAMPLES 2500
		#define UTIL_UPDATE_INTERVAL_MS 1000
		#define UTIL_UPDATE_INTERVAL (UTIL_UPDATE_INTERVAL_MS/STATUS_INTERVAL_MS)
		#define AIRTIME_LONGTERM 3600
		#define AIRTIME_LONGTERM_MS (AIRTIME_LONGTERM*1000)
		#define AIRTIME_BINLEN_MS (STATUS_INTERVAL_MS*DCD_SAMPLES)
		#define AIRTIME_BINS ((AIRTIME_LONGTERM*1000)/AIRTIME_BINLEN_MS)
		bool util_samples[DCD_SAMPLES];
		uint16_t airtime_bins[AIRTIME_BINS];
		float longterm_bins[AIRTIME_BINS];
		int dcd_sample = 0;
		float local_channel_util = 0.0;
		float total_channel_util = 0.0;
		float longterm_channel_util = 0.0;
		float airtime = 0.0;
		float longterm_airtime = 0.0;
		#define current_airtime_bin(void) (millis()%AIRTIME_LONGTERM_MS)/AIRTIME_BINLEN_MS
	#endif
	float st_airtime_limit = 0.0;
	float lt_airtime_limit = 0.0;
	bool airtime_lock = false;

	bool stat_signal_detected   = false;
	bool stat_signal_synced     = false;
	bool stat_rx_ongoing        = false;
	bool dcd                    = false;
	bool dcd_led                = false;
	bool dcd_waiting            = false;
	long dcd_wait_until         = 0;
	uint16_t dcd_count          = 0;
	uint16_t dcd_threshold      = 2;

	uint32_t status_interval_ms = STATUS_INTERVAL_MS;
	uint32_t last_status_update = 0;
	uint32_t last_dcd = 0;

	// Status flags
	const uint8_t SIG_DETECT = 0x01;
	const uint8_t SIG_SYNCED = 0x02;
	const uint8_t RX_ONGOING = 0x04;

    // Power management
    #define BATTERY_STATE_DISCHARGING 0x01
    #define BATTERY_STATE_CHARGING 0x02
    #define BATTERY_STATE_CHARGED 0x03
    bool battery_installed = false;
    bool battery_indeterminate = false;
    bool external_power = false;
    bool battery_ready = false;
    float battery_voltage = 0.0;
    float battery_percent = 0.0;
    uint8_t battery_state = 0x00;
    uint8_t display_intensity = 0xFF;
    uint8_t display_addr = 0xFF;
    bool display_diagnostics = true;    
    bool device_init_done = false;
    bool eeprom_ok = false;
    bool firmware_update_mode = false;

	// Boot flags
	#define START_FROM_BOOTLOADER 0x01
	#define START_FROM_POWERON 0x02
	#define START_FROM_BROWNOUT 0x03
	#define START_FROM_JTAG 0x04

#endif

ahedproductions, 2024-02-19:
Cleaned up the shop, totally. Installed fresh. Placed Config.h from Mark in the directory. Tried make prep and make firmware command from jacobeva - result:
Error during build: exit status 1
make: *** [Makefile:106: firmware-genericesp32] Error 1
Same long list of unexpectd declarations, etc. like the one a bit above.

jacobeva, 2024-02-19:
Things have moved into Board.h now, you will probably need to change the name of the file!