markqvist/RNode_Firmware #57: MODEM and TCXO defines

jacobeva, 2024-01-21:
MODEM should not be defined in the Makefile compiler flags, it should ideally be done in Config.h as according to the example on line 314 for the RAK4630. This should determine the modem in use, so that they can be tied to the board config (as that's more logical). It's not currently done as I couldn't figure out how to pull in the MODEM value from Config.h in the various different libs of the program, any ideas on how to solve this Mark? Same with the TCXO define too, currently the logic to differentiate that for the SX1262 is commented out.

markqvist, 2024-01-21:
Agreed. I'm more familiar with entire build system and code structure, so let me just have a look at this and get it sorted around so we can put those values under the other per-board configuration settings.

jacobeva, 2024-02-10:
Resolved.

jacobeva, 2024-02-10:
See commit e093fa6 and PR https://github.com/markqvist/RNode_Firmware/pull/64

closed 2024-02-10 by jacobeva