markqvist/RNode_Firmware #60: RSSI & SNR values on SX1262 may be incorrect

jacobeva, 2024-01-23:
There is currently no way to verify the RSSI & SNR values of the SX1262 are being correctly handled by the driver (to my knowledge), I need to check these values seem reasonable (they probably aren't) and fix them if not. https://github.com/markqvist/RNode_Firmware/issues/59 must be resolved first to allow me to retrieve the RSSI & SNR values if I'm correct.

jacobeva, 2024-01-23:
Values retrieved look normal. For context the radios are about 30cm apart. Here's the output for reference:
`Round-trip time is 798.89 milliseconds over 1 hop [RSSI 68 dBm] [SNR 14.25 dB] [Link Quality 100.0%]`
Closing...

markqvist, 2024-01-27:
RSSI seems quite off. If 68dBm is reaching the modem, it would be receiving more than 6.3 kilowatts on the antenna. Now, I don't know what kind of transceiver you're designing here, but I now expect some dramatically serious equipment :)

Joking aside, there should be an offset value or calculation somewhere in the datasheet.

jacobeva, 2024-01-29:
The RSSI issue has now been fixed. I wasn't adding the offset in the firmware, and on top of that an unsigned integer was being used for a negative value, causing the reported value to be something in the range of 200 (!) once the offset was added. Both are now corrected, but I suspect there is something wrong with the setTXPower function too, which I will be investigating later today. All my changes will be proposed in a PR.

markqvist, 2024-02-08:
Fixed by #62.

closed 2024-02-08 by markqvist