markqvist/RNode_Firmware #69: Robust logging implementation

jacobeva, 2024-02-17:
The current logging implementation basically consists of a couple of `Serial.print`s on certain conditions.

A robust logging implementation similar to that of Meshtastic should be designed so as to allow for much easier debugging on the host side. Perhaps the RNode should send this over the serial interface following a certain set of bytes, which indicates to RNS this is a log message. Then, the message can be logged and more easily debugged.

jacobeva, 2024-02-17:
I say this is an important feature from experience ;)

faragher, 2024-02-17:
While useful, I worry about the amount of memory the strings would take up. I'd also like to suggest this be a toggleable feature which defaults to off to keep serial traffic down. Still, could be quite useful if properly used.

jacobeva, 2024-02-17:
Indeed, `PROGMEM` should be used to minimise this. This should also be toggleable, I agree, whether it should be on or off by default is up for debate I suppose. I personally don't think it would be too hard on serial traffic, especially given most boards operate at `112500` nowadays.

jacobeva, 2024-02-17:
Perhaps a byte in the EEPROM for logging toggling is in order...

jacobeva, 2024-05-20:
Issue will be addressed in the community fork.

closed 2024-05-20 by jacobeva