markqvist/RNode_Firmware #75: Possible CSMA failure at very low datarate, possibly Bluetooth-related

VladislavF, 2024-03-27:
**Describe the Bug**
When attempting to send a message through an RNode device (Lilygo T-beam 433MHz SX1278 firmware 1.7) using the Sideband app on android, messages always fail to send going from "Establishing link" to "waiting for path" to "failed". This is evidenced by the fact that my two test devices will begin transmitting simultaneously as seen by the red LED indicator.

**To Reproduce**
Connect two RNode devices with sideband (at least one using the android client). After announcements, attempt to send a message from a device using the android app. The message will fail.

**Expected Behavior**
Sending a message from the android app should complete successfully.

**Logs & Screenshots**
I don't know how to get to log output from Android, I can dump a PC side log.

**System Information**
- Android 13/Ubuntu 22.04
- Python 3.9
- 0.8.1, 0.8.0, 0.7.9 affected

**Additional context**
This issue is not present when using the Sideband python/pip package. I can run the same radios with two computers (an Ubuntu machine, and an RPi) running the Sideband client without issue. Or even using nomadnet is not a problem either. This issue is somehow specific to the Sideband android app. Similarly, messages sent from the computer to the mobile app will typically succeed. Messages sent from the app always fail.

I have also tried this with the past 3 versions of the app on two different phones.

VladislavF, 2024-03-27:
Very good. Yes it's SX1278 based.

VladislavF, 2024-03-29:
I take it there isn't a way to run with bluetooth on a linux machine? I would try to test this to see if it's connected to bluetooth, but the reticulum interface options for LoRA seem to be serial only. 
I did try serial with android and I actually have the same problem as with bluetooth.
I used a usb-C dock to connect the rnode via wire and there still seems to be some sort of CSMA issue. perhaps this is an android problem?

faragher, 2024-03-29:
You can connect via Bluetooth and use the port as normal. If /dev/ttyACM0 is USB or Bluetooth doesn't matter to Reticulum, so if you know what the port is named, you can use it as normal.

VladislavF, 2024-03-29:
I had no idea it was possible to run a serial connection over bluetooth... You learn something every day! 
one /dev/rfcomm0 device later I can confirm that this too works properly. Messages flow seamlessly both ways.
Again pointing to the issue being somehow connected to either android or the android app.

VladislavF, 2024-03-29:
FWIW here's a loramon log from my Ubuntu machine with a remote device trying to send a message. Here is what would otherwise be a success (RPi via BT):

```
loramon /dev/ttyACM1 -C -W loracapture --freq 433000000 --bw 62500 --sf 12 --cr 8
[2024-03-28 21:15:22] Opening serial port /dev/ttyACM1...
[2024-03-28 21:15:25] RNode connected
[2024-03-28 21:15:25] Firmware version: 1.70
[2024-03-28 21:15:25] Radio reporting frequency is 433.0 MHz
[2024-03-28 21:15:25] Radio reporting bandwidth is 62.5 KHz
[2024-03-28 21:15:25] Radio reporting TX power is 2 dBm
[2024-03-28 21:15:25] Radio reporting spreading factor is 12
[2024-03-28 21:15:25] Radio reporting coding rate is 8
[2024-03-28 21:15:25] RNode in LoRa promiscuous mode and listening
[2024-03-28 21:15:40] [-132 dBm] [SNR 1.0 dB] [84 bytes]	b'\x00\x02\x00\x06\xdef\xbb\x99v\xb8\x8d$fvF\xc4\xf5\xd8\x8b\x00L\xcb\xb8\xc6.\xa4\x06\x1eR\x88\x02\x8bk\xbe^A\x86XA\xf0\xfb9j\x17\x0fc\xc9"\x91\xd4\xd8/\x17\x17\x19\xb7\xf6\xf0\xadC\x81\xb7P\x90\xda\xf1 5\xb8\xe8\x13\xcaq\x1f\x1b\xff_\xf9\xcat8\x93\x0b8'
[2024-03-28 21:16:33] [-132 dBm] [SNR 1.5 dB] [52 bytes]	b'\xf0\x08\x00k\x9ff\x01M\x98S\xfa\xab"\x0f\xbaG\xd0\'a\x00\x06\xdef\xbb\x99v\xb8\x8d$fvF\xc4\xf5\xd8\x8bduW\x11i\xa9h\x89\x0c\x9ap_P9\xf9g'
```

Here's one that would fail (Android via BT)
```
loramon /dev/ttyACM1 -C -W loracapture --freq 433000000 --bw 62500 --sf 12 --cr 8
[2024-03-28 20:49:27] Opening serial port /dev/ttyACM1...
[2024-03-28 20:49:30] RNode connected
[2024-03-28 20:49:30] Firmware version: 1.70
[2024-03-28 20:49:30] Radio reporting frequency is 433.0 MHz
[2024-03-28 20:49:30] Radio reporting bandwidth is 62.5 KHz
[2024-03-28 20:49:30] Radio reporting TX power is 2 dBm
[2024-03-28 20:49:30] Radio reporting spreading factor is 12
[2024-03-28 20:49:30] Radio reporting coding rate is 8
[2024-03-28 20:49:30] RNode in LoRa promiscuous mode and listening
[2024-03-28 20:49:47] [-131 dBm] [SNR -5.75 dB] [84 bytes]	b'\xe0\x02\x00\x06\xdef\xbb\x99v\xb8\x8d$fvF\xc4\xf5\xd8\x8b\x00\x811\x7f\x94\xea\xef\xff\xf3h\xdc\x1d\x1ckD\xbc\x15J\xc1%\x9d\x82\xab\xc9\x9e\xe5}\xe1;\xbd]A)\xdc\xa8\x14\xb5\xff0(2\xbcd\xc9\x86)\xe7\xc8\xb3F|\x03\x8aT\x9e\xf1z\x04\x84fgm\xd95\xdb'
[2024-03-28 20:49:56] [-130 dBm] [SNR -5.5 dB] [52 bytes]	b'0\x08\x00k\x9ff\x01M\x98S\xfa\xab"\x0f\xbaG\xd0\'a\x00\x06\xdef\xbb\x99v\xb8\x8d$fvF\xc4\xf5\xd8\x8b\'i\r\xc1\xb3\xfaN\xffx\xd8J\x97\x9d1\x82@'
```
Don't know if that's any use, not sure if these messages are encrypted or not

VladislavF, 2024-03-29:
Sorry to spam the thread but I did note a possibly unintentional but only tangentially related behavior:
When changing device configuration parameters such as bandwidth or even changing interfaces from Serial to BT for instance, messages will not pass successfully until the device has been reset or restarted after the change.

I.e. if I change the bandwidth on the app and restart the app, it will not work. I must also reset the device. Same is true for the other node. 
This may be expected, but has caught me off guard several times now where it takes me 10 minutes to remember I have to reset the device and that's why messages aren't going through.

markqvist, 2024-03-29:
Thanks for all the detailed information and logs, very helpful.

> not sure if these messages are encrypted or not

Yes, they are encrypted, but just having the packet flow with timestamps is what is helpful in debugging anyways, so thanks for providing it.

> When changing device configuration parameters such as bandwidth or even changing interfaces from Serial to BT for instance, messages will not pass successfully until the device has been reset or restarted after the change.

Hmm, that is interesting. It's intended behaviour that the host application / Reticulum stack opening the RNode interface needs to be restarted for new radio parameters to be applied. It's currently not "hot-configurable", and for new hardware configuration to take effect you must:

- In the case of Sideband, you need to completely shut down the app, and relaunch it.
- In the case of `rnsd` running on a computer, router or server, you need to restart the master RNS instance.

That being said however, you should be able to do *just* that, without having to power-cycle or reset the RNode device. The RNode should take the new configuration parameters when re-configured by the host. If  this does not happen, there may be a bug that was introduced somewhere along the line, because this is definitely not intended behaviour.

VladislavF, 2024-04-01:
I have tried increasing bandwidth to 125, and 250kHz with the same results. However, going up to 500kHz has resulted in successful communication. So perhaps your guess about data rates is correct. Seems somewhere around 0.7kbps things start working correctly.

markqvist, 2024-04-01:
Ok, interesting, thanks for adding. I did find a *potential* issue with the CSMA RNG init on ESP32 and corrected it. It's in the main source repo now. If it is easy for you to build from source, you can pull the latest version from master and run:

```bash
make firmware-tbeam upload-tbeam
```

That will give you the firmware with the fix on the devices, and you could try and see if it makes any difference. If it does, it is indeed a CSMA issue. If not, my bets are on the LoRa modems losing sync due to the lack of TCXO on the boards.

From what I've seen too, the limit also seems to be around the 0.7Kbps you mention, before needing a very stable clock source. At least for longer packets. Smaller data payloads usually seem to work better, which again points to this happening because of clock drift / sync mismatch.

faragher, 2024-04-01:
While it works on Meshtastic, and it's still pretty in the ballpark, I'd like to point out official specs require 50ppm accuracy for SF12. At 433 MHz, I'd expect 65k to usually work and 125k to always work (as far as frequency mismatch goes) so long as the crystals are actually 20ppm. Since these appear to use the same LoRa module as other LilyGOs, I expect they should work.

That's for raw LoRa with the SX series chips. If there's an additional Reticulum requirement, it would be worth discovering.

VladislavF, 2024-04-01:
I have flashed this firmware and I appear to have the same issue at the lower bandwidths.
Again, I would buy the sync issue if meshtastic also had this problem and if the desktop version of sideband had this problem. But neither of these is the case. 
I have even managed to transfer an image at this configuration (62.5khz) over the period of like 15 minutes between an rpi and ubuntu desktop with sideband

markqvist, 2024-04-01:
Thanks a bunch for putting all this effort into it and collecting information. I'll be looking into this and trying to replicate it when I have a chance, but if you figure out anything else in the meantime, please do share it here as well.

markqvist, 2024-05-07:
Could you try this again with the latest versions of RNS (0.7.4) and the RNode firmware (1.71)? A range of improvements were made that could affect this problem too.

markqvist, 2024-05-31:
Assuming fixed in 1.71, closing. If problem persist, please refer to issue in https://github.com/liberatedsystems/RNode_Firmware_CE.

closed 2024-05-31 by markqvist