markqvist/RNode_Firmware #79: LilyGO LoRa T3S3 resets when communicating over serial

dacwe, 2024-07-12:
**Describe the Bug**
After flashing to LilyGO LoRa T3S3 V1.2 (20230720) and resetting the board manually the serial port cannot be opened. 

```
[14:51:39] Please take the board out of flashing mode by momentarily pressing the RESET button.
[14:51:39] Hit enter when this is done.

[14:51:56] Opening serial port /dev/ttyACM0...
[14:51:59] Serial port opened, but RNode did not respond. Is a valid firmware installed?
write failed: [Errno 5] Input/output error
```

Trying to open the port manually by calling `cat /dev/ttyACM0` also resets the board.

**To Reproduce**
1. `rnodeconf --autoinstall --baud-flash`
2. Choose `[9] LilyGO LoRa T3S3` and `[2] 820 - 1020 MHz (with SX1262 chip)`
3. Put the board into flashing mode and wait for flash to finish
4. Reset manually

**Expected Behavior**
Serial communication works and does not reset the board.

**Logs & Screenshots**

```
------------------------------------------------------------------------------
                               Installer Ready

Ok, that should be all the information we need. Please confirm the following
summary before proceeding. In the next step, the device will be flashed and
provisioned, so make sure that you are satisfied with your choices.

Serial port     : /dev/ttyACM0
Device type     : RNode 820 - 960 MHz
Platform        : ESP32
Device MCU      : Espressif Systems ESP32
Firmware file   : rnode_firmware_t3s3.zip

------------------------------------------------------------------------------

Is the above correct? [y/N] y
[14:50:57] Checking firmware file availability...
[14:50:57] Retrieving latest version info from https://unsigned.io/firmware/latest/?v=2.1.3&variant=rnode_firmware_t3s3.zip
[14:50:57] The selected firmware for this board is version 1.72
[14:50:57] Using existing firmware file: rnode_firmware_t3s3.zip for version 1.72
[14:50:57] Verifying firmware integrity...
[14:50:58] 
[14:50:58] Please put the board into flashing mode now, by holding the BOOT or PRG button,
[14:50:58] while momentarily pressing the RESET button. Then release the BOOT or PRG button.
[14:50:58] Hit enter when this is done.

[14:50:59] Decompressing firmware...
[14:50:59] Firmware decompressed
[14:50:59] Flashing RNode firmware to device on /dev/ttyACM0
esptool.py v3.3.3
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:54:75:db:b3:0c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00010000 to 0x000f4fff...
Flash will be erased from 0x00210000 to 0x003effff...
Flash will be erased from 0x00008000 to 0x00008fff...
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 749.0 kbit/s)...
Hash of data verified.
Compressed 15104 bytes to 10430...
Wrote 15104 bytes (10430 compressed) at 0x00000000 in 0.2 seconds (effective 513.6 kbit/s)...
Hash of data verified.
Compressed 937600 bytes to 573260...
Wrote 937600 bytes (573260 compressed) at 0x00010000 in 8.5 seconds (effective 881.0 kbit/s)...
Hash of data verified.
Compressed 1966080 bytes to 1782227...
Wrote 1966080 bytes (1782227 compressed) at 0x00210000 in 18.3 seconds (effective 858.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 134...
Wrote 3072 bytes (134 compressed) at 0x00008000 in 0.0 seconds (effective 551.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
[14:51:32] Done flashing
[14:51:32] Waiting for ESP32 reset...
[14:51:39] 
[14:51:39] Please take the board out of flashing mode by momentarily pressing the RESET button.
[14:51:39] Hit enter when this is done.

[14:51:56] Opening serial port /dev/ttyACM0...
[14:51:59] Serial port opened, but RNode did not respond. Is a valid firmware installed?
write failed: [Errno 5] Input/output error
```

**System Information**
- Ubuntu 22.04.3
- Python 3.10.12
- rns 0.7.5

dacwe, 2024-07-12:
Tried with two different boards. Same result, both v1.2 but different (production?) dates: 20230720 and 20230530.
[image]

dacwe, 2024-07-12:
Tried installing on a second board. Same error, boards have the same version but different (production?) dates: 20230720 and 20230530. 

[image]

faragher, 2024-07-12:
I have never seen that board configuration before. The breakouts next to the battery connector aren't present on the V2.1 or the T3S3. It's possible some changes have been made that affect the firmware (the T-Beam had a power management module changed in a recent revision that needed a new library altogether).

I suggest checking the CE firmware repo, as this is mostly legacy firmware: https://github.com/liberatedsystems/RNode_Firmware_CE

Here's a pertinent topic from that repo:
https://github.com/liberatedsystems/RNode_Firmware_CE/issues/12

Since I'm not a firmware guy, I can't be of much more help, but you should get a pretty quick response in the other repo.

dacwe, 2024-07-12:
Thanks for the quick reply! I will close this one out!

closed 2024-07-12 by dacwe