markqvist/RNode_Firmware #85: RAK4631 intermittent packet corruption

jacobeva, 2024-10-10:
Continued from the issue originally filed here: https://github.com/markqvist/Reticulum/issues/569

It seems that my fix has solved the packet corruption over USB, but the issue is still present on BLE. This is likely because I need to backport my various changes to the BLE functions & parameters (#83), as in its current state in this repo, it is very unreliable, hence the issue has seemingly returned. I don't think it's the same root cause this time, though.

markqvist, 2024-10-10:
Please check my comment on referenced thread as well; there indeed is a rare condition in the resource transfer code that leads to very long transfer timeouts. I will have it fixed soon. That said, there may still be something going on with the BLE stuff, of course. Just wanted to give you a heads up on that.

jacobeva, 2024-10-10:
So my BLE changes haven't fixed the issue. There's still corruption when sending packets over BLE to the RNode. Why, I cannot ascertain at the moment, but I will investigate further. I recommend merging #86 for now as a fix for USB connections.

jacobeva, 2024-10-10:
Looks like my fix has only made the issue occur less often. It's still present currently, it's more apparent when transmitting larger files with an SX1280 modem (like I said before, seems to be a link between higher data rates a higher amounts of corruption). It's not as bad as it was before, though.

jacobeva, 2024-10-10:
Never mind, the SX1280 bug seems to be a separate one entirely. For matter of scope, discussion on this issue will remain on the BLE issue.

PR #94 (2024-10-23): Fix BLE packet corruption on nRF52
Closes #85
closed 2024-10-23 by markqvist