markqvist/RNode_Firmware #92: RAK4361 Failure

ToxicityKingETN, 2024-10-13:
RAK fails to flash RNODE firmware after using rnodeconf, after receiving notification that device was programmed, when it attempts to verify by opening the previously used com port it will either fail, or rnode will fail to respond.
I believe the port fails because when the RAK reboots it does not automatically enter dfu mode, so the port may be different than the one used during the install(i.e. when mine was in DFU, it was COM5, but after reboot it was COM4.
As for the Rnode not responding for verification or to attempt to enable ble, I am not able to tell the reason.

**To Reproduce**
Describe in detail how to reproduce the bug.
[image]

**System Information**
- Windows 11
- Python 3.12

**Additional context**
Keep up the good work and thank you for your efforts!

markqvist, 2024-10-13:
Thanks for reporting! Just calling in @jacobeva on this one too.

ToxicityKingETN, 2024-10-14:
Additional, tried flashing it from my linux machine to see if it was windows reassigning the com port. Still failed.
Here it is as ACM1, after flashing, nRF52 reboots, then changes com port again
[image]
Changed com port
[image]

jacobeva, 2024-10-23:
Yeah, I've had this issue from time to time as well. Sometimes the RAK4631 jumps ports. I can't really say why at the moment, it seems somewhat random. I will take this opportunity to mention again that we should really be targeting devices by `/dev/serial/by-id` in `rnodeconf` ;)

Sorry for taking so long to respond. But I think that's basically the long and short of it. Don't know why it happens, and targeting the device a different way is currently the only way I can think of fixing it (unless this is related to the USB serial in some way, which is possible).

closed 2024-11-03 by ToxicityKingETN