markqvist/RNode_Firmware #95: RAK4631 firmware hash calculation error (Radio state mismatch bug)

jacobeva, 2024-12-06:
An error in the algorithm which calculates the hash of the firmware image on the device causes it to not work properly when the compiled firmware is over a certain size on the RAK4631. This has been known to me for some time, but I've not had the time to make a fix until now.

This is what has been causing the mysterious "radio state mismatch" issue people have been reporting on v1.78 of this firmware.

There are two possible solutions:
- rework the algorithm to get the firmware image size from a header, similar to ESP32 (unknown if possible)
- add an option to rnodeconf to set the firmware image size so the device can use this in the algorithm (already implemented in another repository, but a less elegant solution)

I will investigate the first solution and see if it is possible. If not, the second will be my go-to.

jacobeva, 2024-12-06:
Excellent news! It seems the first option is actually viable. The implementation will take some time, but it should indeed work! I just need to read the first 5 bytes at `0xFF008` in the flash memory of the nRF52840. I should note this will likely have to be redone for any other nRF chip type.

ToxicityKingETN, 2024-12-06:
Thank you for pushing the issue, I was able to resolve using liam cottle's web flasher that allowed me to set correct firmware hash, I realized the previous linked issue it showed it in TNC mode, I should not have shown that because it is misleading. The mismatch was regardless of operating mode. Thank you guys and keep up the amazing work.

jacobeva, 2024-12-07:
Yeah I first noticed this bug a couple days ago when someone DMed me about it. So it's been known about for a couple days. Hence when Mark mentioned me in Liam's repo I immediately clocked it was the same bug :)

jacobeva, 2024-12-07:
#96 fixes this.

commit 5ac30b0f (2024-12-07): Retrieve firmware image length from bootloader cfg, fix #95
commit 9fdd886d (2024-12-07): Merge pull request #96 from jacobeva/hash-fix

Retrieve firmware image length from bootloader cfg, fix #95
closed 2024-12-07 by markqvist