markqvist/RNode_Firmware #97: Heltec T114

cvetaevvitaliy, 2025-01-01:
I see the `Heltec T114` device in the flasher https://liamcottle.github.io/rnode-flasher/
but I can't find the firmware in the release

markqvist, 2025-01-01:
Support was added after the last release (v1.79). @liamcottle is finishing up the last stuff for the firmware. Once it's ready, he'll submit a pull request, I'll merge it and release the compiled firmware here.

markqvist, 2025-01-02:
PR is in: https://github.com/markqvist/RNode_Firmware/pull/98?notification_referrer_id=NT_kwDOABXxPLMxNDA4NjM3OTcxODoxNDM4MDEy

cvetaevvitaliy, 2025-01-02:
> PR is in: https://github.com/markqvist/RNode_Firmware/pull/98

Awesome. I have HW, I can test it

cvetaevvitaliy, 2025-01-02:
Flashed, works 

[image]

markqvist, 2025-01-17:
Full support added in version `1.81`.

closed 2025-01-17 by markqvist