markqvist/RNode_Firmware #105: Selected device is not an RNode!

MarcusRM44, 2025-02-12:
Following Andy Kirby's video exactly in a Chrome browser.
Flashes OK.
When I "Provision" I get the error message "Selected device is not an RNode!"
Tried many times.

LinuxinaBit, 2025-02-13:
It’s possible serial is somehow disabled on your OS…
Can you provide some info like specific OS/version and browser installation method?

KenSamson, 2025-02-13:
You may need to hit the reset button after flashing, and the port may
change.   In cases where I've done that, I've been able to move forward.

MarcusRM44, 2025-02-13:
I don't understand the question about serial disabled.

Chrome...

Chrome is up to date
Version 133.0.6943.99 (Official Build) (64-bit)

Otherwise, I am just running a regular installation of Windows 11.

Marcus

LinuxinaBit, 2025-02-14:
When you click `Enter DFU Mode` is there a popup and what does it say?

vitz313, 2025-02-18:
I have the same issues with my **Heltec32V3**, using chrome version **133.0.6943.99** (Official Build) (64-bit) in **windows 10** Pro 22H2. I am able to successfully flash the Heltec, but when I try to provision it I get "Selected device is not an RNode!"

When I read the serial on the Heltec this is what I get...

07:55:33.657 -> ESP-ROM:esp32s3-20210327
07:55:33.657 -> Build:Mar 27 2021
07:55:33.657 -> rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
07:55:33.657 -> SPIWP:0xee
07:55:33.657 -> mode:DIO, clock div:1
07:55:33.657 -> load:0x3fce3808,len:0x4bc
07:55:33.657 -> load:0x403c9700,len:0xbd8
07:55:33.657 -> load:0x403cc700,len:0x2a0c
07:55:33.657 -> entry 0x403c98d0
07:55:33.922 -> E (272) esp_core_dump_flash: Incorrect size of core dump image: -200799214
07:55:34.203 -> ⸮U⸮⸮Device unprovisioned, no device configuration found in EEPROM
07:55:44.786 -> ⸮'

[image]

[image]

LinuxinaBit, 2025-02-18:
@liamcottle Is this a known issue on your end?

MarcusRM44, 2025-02-20:
Interesting that your Win 10 doesn't work. I couldn't get the provisioning on Win 11 to work, so I created a virtual Win 10 on the same PC using VMWare.
Result success.

Marcus

screen1.jpg (view on web)<https://github.com/user-attachments/assets/c0707733-791e-4a79-9a71-f3dc3a0a2650>

screen2.jpg (view on web)<https://github.com/user-attachments/assets/560eecff-1b60-4aec-8f2c-44c4521dabd4>

screen1.jpg (view on web)<https://github.com/user-attachments/assets/c0707733-791e-4a79-9a71-f3dc3a0a2650>

screen2.jpg (view on web)<https://github.com/user-attachments/assets/560eecff-1b60-4aec-8f2c-44c4521dabd4>

markqvist, 2025-02-20:
Closing this since it's not related to the firmware itself, but to the web-flasher.

closed 2025-02-20 by markqvist