markqvist/RNode_Firmware #111: Error flashing latest firmware to RAK4631

cwlls, 2025-06-03:
**Describe the Bug**
When using the `rnodeconf -a /dev/...` command the flashing proceeds through each step, but fails in the EEPROM validation step, leaving the device in an unusable state. This is true with versions 1.82 and 1.81. Flashing with version 1.80 works as anticipated.

**To Reproduce**
Attempt to flash a RAK4631 device using firmware versions 1.82 or 1.81

**Expected Behavior**
The flashing process should succeed leaving the device in a functional, albeit "experimental" state (given the current status of the RAK4631 within RNode)

**Logs & Screenshots**
```
------------------------------------------------------------------------------
                               Installer Ready

Ok, that should be all the information we need. Please confirm the following
summary before proceeding. In the next step, the device will be flashed and
provisioned, so make sure that you are satisfied with your choices.

Serial port     : /dev/cu.usbmodem12101
Device type     : RAK4631 779 - 928 MHz
Platform        : NRF52
Device MCU      : Nordic Semiconductor nRF52840
Firmware file   : rnode_firmware_rak4631.zip

------------------------------------------------------------------------------

Is the above correct? [y/N] y
[11:40:22] Checking firmware file availability...
[11:40:22] Retrieving latest version info from https://unsigned.io/firmware/latest/?v=2.4.0&variant=rnode_firmware_rak4631.zip
[11:40:24] The selected firmware for this board is version 1.82
[11:40:24] Using existing firmware file: rnode_firmware_rak4631.zip for version 1.82
[11:40:24] Verifying firmware integrity...
[11:40:25] Decompressing firmware...
[11:40:25] Firmware decompressed
[11:40:25] Flashing RNode firmware to device on /dev/cu.usbmodem12101
Upgrading target on /dev/cu.usbmodem12101 with DFU package /Users/cwlls/.config/rnodeconf/update/1.82/rnode_firmware_rak4631.zip. Flow control is disabled, Dual bank, Touch 1200
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
##################################
Activating new firmware
Device programmed.
[11:40:53] Done flashing
[11:40:53] Waiting for NRF52 reset...
[11:40:59] Opening serial port /dev/cu.usbmodem12101...
[11:41:02] Device connected
[11:41:02] Current firmware version: 1.82
[11:41:02] Reading EEPROM...
[11:41:03] Clearing old EEPROM, this will take about 15 seconds...
[11:41:26] Waiting for NRF52 reset...
[11:41:44] Loading signing key...
[11:41:44] Bootstrapping device EEPROM...
[11:41:50] EEPROM written! Validating...
[11:41:50] Setting firmware checksum...
[11:41:57] Waiting for NRF52 reset...
[11:42:18] EEPROM was written, but validation failed. Check your settings.
```

**System Information**
- OS and version: macOS 15.3.2 (24D81)
- Python version: Python 3.13.2 (running within a venv)
- Program version: rnodeconf 2.4.0

**Additional context**
Meshtastic was previously installed on this device, but after a successful flash to version 1.80 of RNode, the issues persist.

oherrala, 2025-09-13:
Same issue here with Heltec T114 with firmware version 1.83:

```
------------------------------------------------------------------------------
                               Installer Ready

Ok, that should be all the information we need. Please confirm the following
summary before proceeding. In the next step, the device will be flashed and
provisioned, so make sure that you are satisfied with your choices.

Serial port     : /dev/cu.usbmodem2101
Device type     : Heltec Mesh Node T114 850 - 950 MHz
Platform        : NRF52
Device MCU      : Nordic Semiconductor nRF52840
Firmware file   : rnode_firmware_heltec_t114.zip

------------------------------------------------------------------------------

Is the above correct? [y/N] y
[20:46:12] Checking firmware file availability...
[20:46:12] Retrieving latest version info from https://unsigned.io/firmware/latest/?v=2.4.0&variant=rnode_firmware_heltec_t114.zip
[20:46:12] The selected firmware for this board is version 1.83
[20:46:12] Using existing firmware file: rnode_firmware_heltec_t114.zip for version 1.83
[20:46:12] Verifying firmware integrity...
[20:46:13] Decompressing firmware...
[20:46:13] Firmware decompressed
[20:46:13] Flashing RNode firmware to device on /dev/cu.usbmodem2101
Upgrading target on /dev/cu.usbmodem2101 with DFU package /Users/oherrala/.config/rnodeconf/update/1.83/rnode_firmware_heltec_t114.zip. Flow control is disabled, Dual bank, Touch 1200
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
#######
Activating new firmware
Device programmed.
[20:46:43] Done flashing
[20:46:43] Waiting for NRF52 reset...
[20:46:49] Opening serial port /dev/cu.usbmodem2101...
[20:46:52] Device connected
[20:46:52] Current firmware version: 1.83
[20:46:52] Reading EEPROM...
[20:46:52] Clearing old EEPROM, this will take about 15 seconds...
[20:47:05] Waiting for NRF52 reset...
[20:47:24] Loading signing key...
[20:47:24] Bootstrapping device EEPROM...
[20:47:29] EEPROM written! Validating...
[20:47:30] Setting firmware checksum...
[20:47:37] Waiting for NRF52 reset...
[20:47:58] EEPROM was written, but validation failed. Check your settings.
```

jedahan, 2025-10-01:
Same with RAK4631
```
------------------------------------------------------------------------------
                               Installer Ready

Ok, that should be all the information we need. Please confirm the following
summary before proceeding. In the next step, the device will be flashed and
provisioned, so make sure that you are satisfied with your choices.

Serial port     : /dev/cu.usbmodem101
Device type     : RAK4631 779 - 928 MHz
Platform        : NRF52
Device MCU      : Nordic Semiconductor nRF52840
Firmware file   : rnode_firmware_rak4631.zip

------------------------------------------------------------------------------

Is the above correct? [y/N] y
[17:59:15] Checking firmware file availability...
[17:59:15] Retrieving latest version info from https://unsigned.io/firmware/latest/?v=2.4.0&variant=rnode_firmware_rak4631.zip
[17:59:15] The selected firmware for this board is version 1.83
[17:59:15] Using existing firmware file: rnode_firmware_rak4631.zip for version 1.83
[17:59:15] Verifying firmware integrity...
[17:59:16] Decompressing firmware...
[17:59:16] Firmware decompressed
[17:59:16] Flashing RNode firmware to device on /dev/cu.usbmodem101
Upgrading target on /dev/cu.usbmodem101 with DFU package /Users/micro/.config/rnodeconf/update/1.83/rnode_firmware_rak4631.zip. Flow control is disabled, Dual bank, Touch 1200
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
########################################
##################################
Activating new firmware
Device programmed.
[17:59:45] Done flashing
[17:59:45] Waiting for NRF52 reset...
[17:59:51] Opening serial port /dev/cu.usbmodem101...
[17:59:54] Device connected
[17:59:54] Current firmware version: 1.83
[17:59:54] Reading EEPROM...
[17:59:55] Clearing old EEPROM, this will take about 15 seconds...
[18:00:18] Waiting for NRF52 reset...
[18:00:36] Loading signing key...
[18:00:36] Bootstrapping device EEPROM...
[18:00:42] EEPROM written! Validating...
[18:00:42] Setting firmware checksum...
[18:00:50] Waiting for NRF52 reset...
[18:01:10] EEPROM was written, but validation failed. Check your settings.
```

markqvist, 2025-11-26:
> `[2025-11-26 00:26:59] [Debug] RNodeInterface[Interface LoRa1] Radio reporting state is offline`

The radio refuses to turn on. This is most likely because some step of the boot checks failed, like the firmware hash being incorrect, or the device not having a correct configuration present in EEPROM.

If you put the device into DFU mode manually when flashing, the actual firmware flashing might proceed fine, but as soon as `rnodeconf` tries to restart the device and load the config, set the firmware hash, etc., the device might stay in DFU mode, and `rnodeconf` can't communicate with the firmware.

If there's no display on the device, you're not seeing the "Missing config" or "Firmware corrupt" error messages, which can make it less obvious, but the `rnodeconf -a` output has likely given you an error while flashing about some steps failing.

If you really have to put the device into DFU mode *manually*, you will also need to press the reset button manually once `rnodeconf` says something like "Waiting for device reset...".

But ideally, you shouldn't have to put it into DFU mode yourself, the flasher should handle it so it can properly reset the device when needed.

markqvist, 2025-11-28:
> `module 'hashlib' has no attribute 'file_digest'`

That's the problem. It can't calculate and set the firmware hash. What Python version and operating system are you on?