markqvist/RNode_Firmware #113: Bluetooth on Android 15 Not Working With RNode FW 1.83

weltamdraht, 2025-08-04:
**Describe the Bug**
I installed the current RNode FW 1.83 on:
TLORA T3S3
T-Echo
Heltec V3

Tried to connect those three RNodes using bluetooth to:
a SONY XPERIA compact G8441 Android 9 
a ThinkCentre Desktop PC
a Samung Xcover Pro 6 Android 15

Also also tried a Samsung Galaxy A55 running Android 15 and a recent iphone(Apple).

1) SONY XPERIA compact G8441 Android 9:
All RNodes can be seen advertising. 
All RNodes can be connected and work fine.
(using Sideband 1.7.0)
2) Desktop PC (Debian 12):
Advertisement of all RNodes are seen.

3)  Samung Xcover Pro 6 Android 15 (plus Galaxy A55 and a fairly recent iphone):
RNodes do not show up. 
Neither in Androids bluetooth menu nor using Sideband ...

Flashed the ,TLORA T3S3, T-Echo, Heltec V3 with meshtastic firmware (just for testing purposes): Everything works fine. All devices show up and can be paired.

**To Reproduce**
A recent Android 15 phone and one of the LoRa devices mentioned above.

**Expected Behavior**
RNodes should be usable with bluetooth like on the SONY XPERIA compact G8441 Android 9. 

**Logs & Screenshots**
Please - I could make a youtube Video but I don't want to :-) 
Please let me know if I can provide any logs or traces. 

**System Information**
Debian12, Android 9, Android 15, Sideband 1.7.0, RNode 1.83, rns 1.00, Python 3.11.2

**Additional context**
I really enjoy tinkering around with reticulum an admire all the work that went into it. It's a great project I really honestly enjoy. Thank you for making that possible.

multikatt, 2025-09-06:
Ill make this matrix post i made here as well:

> Found this about meshtastic: https://github.com/meshtastic/firmware/issues/6958 and enable and enter the DEVELOPER OPTIONS of your phone and enable "Show unsupported Bluetooth LE devices in Settings"
> 
> Tried that out and now its working for me. However, this meshtastic issue is just simply closed with "Alas, we weren't able to find a way to fix this :(" but it seems to be working fine with android 15 and BT so i guess they found a solution elsewhere

weltamdraht, 2025-09-10:
@multikatt 
Thank you!!!! This works fine. 
Will stop carrying my laptop around for experiments then :-)

Will close this issue then since I don't have an iPhone to test with and nobody seems to want to solve it in the sources.

weltamdraht, 2025-09-10:
Closed - seems to be the same issue as in #112