markqvist/RNode_Firmware #115: Explicit SyncWord definition

racerxdl, 2025-08-26:
I was thinking about creating a discussion about it, but its technically a bug.

LoRa sadly uses the SyncWord in the preamble to determine the network thats currently receiving. RNode does not set that explicity, so it defaults to the hardware default, which can make __some__ devices has some communication issues.

The thing is that, the default for SX devices seens to be 0x34  which is defined as Public LoRaWAN. There is also 0x12 for "private" LoRaWAN.

That sync word is also used as a hardware filter for the PHY, so using a common LoRaWAN ID might keep RNode awake more time than needed by parsing packages that are not rnode compatible.

> My suggestion is to set a non 0x34 default value for RNode, so we don't receive trash packets from common LoRa devices.

Just FYI: I discovered this because I was trying to make a parser that works for both meshtastic and rnode packets (seens like we can differentiate them), and when I tried to receive a meshtatistc packet with rnode, it didnt worked. Then I found out it uses the default `0x2B` as SyncWord.

The main downside of my suggestion is that it will make it unable to communicate with devices that has 0x34 as their syncword 😢 

PS: If needed, I can make a PR with the required changes. It seens like the setSyncWord is already in the code, just not being called anywhere. Maybe we could also add it to the API so it can be set at the host side instead hardcoded.

XDjackieXD, 2025-12-09:
@fr33n0w LoRa Sync Words are 2 Byte long and RNode sets it to 0x1424 which is one of the two standard sync words (the one for private networks).
The other one is 0x3444 (which is specified for public LoRaWAN networks). For older LoRa chips with 1 Byte sync word config the mapping is 0xUV -> 0xU4V4 (so the 1-Byte Version of the sync word used by RNode is 0x12).

XDjackieXD, 2025-12-09:
Is your sending device also a sx126x or a sx127x? I'm currently trying to figure out what could cause this.
While it is expected that receiving with 0x42 set works too (see the blog post I linked at the top), it should *also* work with 0x12. I just checked both your code and the RNode code (are you using this one or the community edition?) and you are both setting the syncword correctly for 0x12 so it should be working :/